realsenseai/librealsense #8111: capture depth via opencv videoreader

eyebies, 2021-01-05:
Camera Model : D400
OS : Ubuntu 20

### Issue Description
Hi - I am trying to access the depth using opencv API. I am able to access the ir feed and the rgb feed using the cv2.VideoCapture ! but I am facing issue when I try to open the depth device 
/tmp/pip-req-build-vu_aq9yd/opencv/modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video4): can't open camera by index
v4l2 is able to list the format of the device !I am unable to view the feed on guvcview as well. The realsense api works fine.

MartyG-RealSense, 2021-01-05:
Hi @eyebies  I hope that the advice in the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/6841#issuecomment-660859774

eyebies, 2021-01-05:
thanks a lot !

eyebies, 2021-01-05:
https://docs.opencv.org/4.4.0/db/d08/tutorial_intelperc.html

closed 2021-01-05 by eyebies