realsenseai/librealsense #8112: Frame timestamp discrepancy

jasdeep06, 2021-01-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 2.39.0.2337 | 
| Operating System & Version | Win10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | python| 
| Segment			|other| 

### Issue Description
I am trying to extract latest frame using wait_for_frames/poll_for_frames but the frame timestamp that I am getting using get_frame_timestamp has a lower value when compared to the timestamp at the time of calling wait_for_frames/poll_for_frames.

I am using the following code for the same.

pipeline=rs.pipeline()
config=rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 6)
profile=pipeline.start(config)
d_sensor=profile.get_device().first_depth_sensor()
d_sensor.set_option(rs.option.visual_preset, 3)
depth_scale = d_sensor.get_depth_scale()
d_sensor.set_option(rs.option.global_time_enabled,1)
c=0
    try:
        while c<5:
            t1=time.time()
            print(t1)
            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            if not depth_frame:
                continue
            d_time = depth_frame.get_timestamp()/1000
            print("Depth frame timestamp:"+str(d_time))
            depth_image = np.asanyarray(depth_frame.get_data())/depth_scale
            c+=1
 finally:
        pipeline.stop()

Output:

1609860437.379545
Depth frame timestamp:1609860437.3582704
1609860437.5453537
Depth frame timestamp:1609860437.5248868
1609860437.7119782
Depth frame timestamp:1609860437.6915023
1609860437.8788202
Depth frame timestamp:1609860437.858115
1609860438.045043
Depth frame timestamp:1609860438.0247257

MartyG-RealSense, 2021-01-05:
Hi @jasdeep06   Do you get more satisfactory timestamp results if you set **rs.option.global_time_enabled** to '0' (false) instead of '1' (enabled) please?

I ask this because I recall a recent case where drift between timestamps was significantly reduced when global time was made false.

https://github.com/IntelRealSense/librealsense/issues/8003

ev-mp, 2021-01-05:
@jasdeep06 hello, this is a possible and even a plausible scenario as the frame's timestamp designates either (a) the time point the frame was generated in Hardware or (b) the time it was acquired by the SDK. 
In first case it should occur 100% of the time, while in the latter - it depends on the timestamp type and the propagation latency, still for most cases the timestamp should be lower than the time `wait_for_frames` call was made.
Please see the [documentation](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1) on timestamp types

closed 2021-01-19 by MartyG-RealSense