realsenseai/librealsense #8114: Imgui integration

arothenberg, 2021-01-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
I'm trying to create some widgets. Specifically an indicator using bitmaps that rotate...
So far I have used the thirdparty(folder) IMGUI and it's worked great. Everything is running smoothly.
I'm happy with your product. 

But since  I need to create some graphical indicators to represent calculations on my depth table I've had to download the latest Imgui -  since the examples I've seen didn't compile in the realsense environment. 

I noticed that last week a similar question, in Issues, was posed and the answer was that the version of Imgui currently  being used by realsense worked fine. Do I  need to figure out ways  integrate solutions I find from ocornut(mainly)  with the realsense environment or is there a simpler way to do what I want? 

If you would like specifics I could paste some code and errors. 

Thanks.

MartyG-RealSense, 2021-01-06:
Hi @arothenberg   The ideal situation may be to try to create your graphical elements using the SDK's existing Imgui implementation.   The discussion in the link below has directions to information about the SDK's Imgui draw commands and advice about custom rendering.

https://support.intelrealsense.com/hc/en-us/community/posts/360052500553-How-to-draw-shapes-on-a-window-using-ImGiu-

arothenberg, 2021-01-06:
I'll try looking closer at the Imgui.h procedures and see if it will cover my needs. There is code for rotating bitmaps ocornut uses , as an example, that has a function that performs the rotation but that function is absent in your version. 

Alternately, I could try and write my own  code using a rgba8 type pointer  and use glTexImage2D  to render it - I'll try that if there is no easy way through your Imgui implementation to achieve my needs.

Thanks.

MartyG-RealSense, 2021-01-06:
Okay, thanks very much for the update.   Good luck!

closed 2021-01-20 by MartyG-RealSense