realsenseai/librealsense #8118: D455 IMU Calibration Error

robotsoft, 2021-01-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version | Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-59                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.40.0                          | 
| Language                            |  python                        | 
| Segment			|  Robot                         | 

### Issue Description
When I tried to run rs-imu-calibration.py according to https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration, I encountered the following error message.

```
Start interactive mode:

Error: Enumeration keys must be strings: (0,)
```

MartyG-RealSense, 2021-01-06:
Hi @robotsoft   This error message does not only occur with librealsense but with other programs that involve the installation of **enum**.  Typically, the message occurs if **enum34** was not used as the enum installation instruction.

The Ubuntu instructions for enum apparently have a layout error where the enum instruction is bolded as headline text. 

[image]

I believe that it should read as follows:

Enum
Python 2.7 
**sudo pip install enum34**

robotsoft, 2021-01-06:
It works. Thank you, @MartyG-RealSense.

MartyG-RealSense, 2021-01-07:
Great news @robotsoft  - thanks for the update!

closed 2021-01-20 by MartyG-RealSense