realsenseai/librealsense #8119: how to combine a depth frame and a color frame into a frameset

msh1020, 2021-01-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |05.12.09.00 & 05.11.01.100 | 
| Operating System & Version |   Win 10  |
| Platform                                 | PC |
| SDK Version                            |  2.13.0 }                          | 
| Language                            | python}                          | 
| Segment			| Robot | 

### Issue Description
<Describe your issue / question / feature request / etc..>
     when I use two D415 at the same time, I get the color frame and depth frame by 'device_manager.poll_frames()'，but how can I combine the color frame and the depth frame into a frameset?
    Now my code is as follows:

       frames = device_manager.poll_frames()
        for (device, frame) in frames.items():
            print('frame:', frame)
            frame = frame.as_frameset()

    Error is as follows:
    AttributeError: 'dict' object has no attribute 'as_frameset'

MartyG-RealSense, 2021-01-06:
Hi @msh1020   Does the information in the link below help to answer your question, please?

https://github.com/IntelRealSense/librealsense/issues/5847

msh1020, 2021-01-12:
> Hi @msh1020 Does the information in the link below help to answer your question, please?
> 
> #5847

Thanks, it works.

MartyG-RealSense, 2021-01-12:
Great news that it worked for you @msh1020 - do you require further assistance with this case, please?   Thanks!

closed 2021-01-19 by MartyG-RealSense