realsenseai/librealsense #8120: cmake Error while installing Unity wrapper on Ubuntu 18.04

PattyUnity, 2021-01-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 and D435i | 
| Firmware Version                   | v2.41.0 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-58-generic                                      | 
| Platform                                 | PC amd64  |
| SDK Version                            | v2.41.0                          | 
| Language                            |  unity                           | 
| Segment			|  Robot/MR                      | 

### Issue Description
Hi I'm a new user trying to do SLAM with RealSense in Unity on Ubuntu 18.04. I have installed Unity (2019.4.0f1) on Ubuntu. 
I have installed RealSense SDK on Linux by following https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

I then follow https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity for Unity wrapper installation. 
When I execute...
```
cmake .. -DBUILD_CSHARP_BINDINGS=ON -DBUILD_UNITY_BINDINGS=ON -DBUILD_SHARED_LIBS=ON \
-DDOTNET_VERSION_LIBRARY=3.5 -DCMAKE_GENERATOR_PLATFORM=x64 \
-DUNITY_PATH=</home/ril/Unity/Hub/Editor/2019.4.0f1/Editor>Unity
```
I got this error...
```
Cmake Error aat CMakeList.txt:3 (project):
	Generator
	Unix Makefiles
	does not support platform specification, but platform
	X64
	was specifiled
Cmake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
Cmake Error: CMAKE_C_COMPILER not set, after EnableLanguage
```

I searched online for solution of the same issue but it doesn't help me. I have tried the following:
1. Installing Visual Studio Code. (source: https://xspdf.com/help/52803626.html)

2. Install .NET

    -- The error is still the same --

3. Update cmake to the latest version (source: https://github.com/CREDITSCOM/node/issues/2)
Execute `sudo apt-get install cmake` but nothing has changed since my cmake version is 3.10.2 and it states it's the latest version.

4. Tried to install cmake compiler
Execute ` sudo apt-get install g++` and `sudo apt-get update && sudo apt-get install build-essential` but no change because all has been installed

5. I couldn't find any relevant issues in this github. Is it possible to use RealSense in Unity on Ubuntu? My ultimate goal is to use Unity to bridge Hololens together with RealSense (probably on robot) to do SLAM. Suggestions are welcome.

Thanks for your help!

RealSenseSupport, 2021-01-22:
Hi,

What cmake generator are you using? Ninja? I've had more success when the C# bindings on with this one for example:

https://cmake.org/cmake/help/latest/generator/Visual%20Studio%2016%202019.html

WenqiangX, 2021-01-26:
Hi, I met the same issue, can you give us a straight answer **Is it possible to use RealSense in Unity on Ubuntu?** @RealSenseSupport 

Thanks!

RealSenseSupport, 2021-02-15:
Hi,

Will you be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

WenqiangX, 2021-02-15:
I did not receive any feedback regarding my question, I think neither the author who post this issue. Could you please provide any insightful feedback? @RealSenseSupport

PattyUnity, 2021-02-18:
> Hi,
> 
> What cmake generator are you using? Ninja? I've had more success when the C# bindings on with this one for example:
> 
> https://cmake.org/cmake/help/latest/generator/Visual%20Studio%2016%202019.html

Hi, 
The cmake generator in used is UNIX. 

The Visual Studio you mention is the IDE, not Visual Studio Code, am I right? From my limited knowledge, Visual Studio IDE is not available in Ubuntu, only VS code is. If this is the case, does it mean that Realsense Unity wrapper isn't supported in Ubuntu? 

Thanks.

closed 2021-09-17 by MartyG-RealSense