realsenseai/librealsense #8121: How to get the data of edge point?

junhui519, 2021-01-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i } | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version | Linux (Ubuntu 18.04)  ROS | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | PC |
| SDK Version                            |  latest | 
| Language                            | C++/py                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, I'm working to design a robot with wheels which climb up a box. So now I'm trying to detect the box and measure the distance or coordinate of the edges of the front face of the box to change the direction of the robot. What should I do or refer to?

RealSenseSupport, 2021-01-08:
@junhui519 Please check if below two examples help you or not.
https://github.com/IntelRealSense/librealsense/tree/master/examples/measure 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

junhui519, 2021-01-11:
Actually, what I want to do is detect an obstacle. So I'm trying to use show_center_depth.py. Is there anything that I can refer to?

RealSenseSupport, 2021-01-12:
@junhui519 Sorry we don't have such examples. I think you might search other projects which focus on Collision/obstacle avoidance.

RealSenseSupport, 2021-01-26:
@junhui519 Any other questions for this ticket? Thanks!

RealSenseSupport, 2021-02-03:
@junhui519 Sorry that we don't hear from you for weeks, will close it at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2021-02-03 by RealSenseSupport