realsenseai/librealsense #8122: D415 on/off wear

BGA3, 2021-01-06:
At my company, we are using the D415 for a given project. We want to take one depth image every 10th second (+/- 1 sec ish), everyday.
Now, we also would like to minimize power usage so we do not want the camera/emitter to  be turned on constantly. Does anyone know, if it would (over time) damage the camera to be turned on / off every 10 seconds for several years?

MartyG-RealSense, 2021-01-06:
Hi @BGA3    Advice on laser degradation over time and the effects on depth quality is provided in the link below by Dorodnic the RealSense SDK Manager.

https://github.com/IntelRealSense/librealsense/issues/3452#issuecomment-471492320

As RealSense 400 Series cameras can operate indefinitely so long as they remain within recommended temperature tolerances and they are also power-efficient, it may be easier to simply leave the camera powered constantly if the gap between captures is only 10 seconds.

BGA3, 2021-01-06:
@MartyG-RealSense  okay, so there is no degradation of the vision processor over time if running continously for years? and if I understand you, I should atleast turn on/off the emitter to avoid degradation.
Do you know if the vision-processor will face issues if also turned on/off?

MartyG-RealSense, 2021-01-06:
At the time of writing this it has been three years since the 400 Series launched and the number of times that problems have been suspected to be related to hardware failure (whether the Vision Processor D4 or other components such as the camera module) have been very rare.  Previous generations of RealSense camera hardware have also been highly resilient and reliable over a multi-year period.

You can of course implement an on-off toggle mechanism if you believe that it is important to the long-term health of your product's camera hardware.  Rather than turn the laser off completely though, you could instead change the **Laser Power** setting, putting it at a value of '1' during inactivity to keep the projector active but low-powered and then set it to '150' (the default Laser Power value) prior to a capture.  Then return its value to '1' until the next capture.

The effect of minimizing Laser Power on the depth image will likely be that there is no visible depth image until the value is increased prior to capture.

BGA3, 2021-01-06:
Thanks for your input, its greatly appreciated!
I am not worried about the cameras vision processor, the benefit from turning the camera completely off would be that we use less power as the camera is turned off for ~90% of the time. And as I understand you, the camera should not be damaged by turning on/off all the time, or am I wrong?
and also, would it be less harmful for the emitter to be alternating between power = "1" and power = "150" rather than on/off?
Thanks in advance

MartyG-RealSense, 2021-01-06:
I would be more concerned about the long-term health of the projector if it were undergoing continuous 'stress test' conditions (being toggled multiple times per second) over a long period.  Ultimately, I don't think that toggling on-off every 10 seconds or changing the Laser Power value is going to have an impact on the camera hardware's physical health.  If you believe that it is the right choice for your paroject then by all means implement it.

MartyG-RealSense, 2021-01-13:
Hi @BGA3  Do you require further assistance with this case, please?   Thanks!

BGA3, 2021-01-13:
Hi @MartyG-RealSense , no thanks I have what i need. I will proceed by turning the emitter off,  waitt 10sec and then turn it on when I need to acquire an image. The camera will be running continously.

Many thanks for your support!

closed 2021-01-13 by BGA3

MartyG-RealSense, 2021-01-13:
I'm very pleased to hear that you reached a satisfactory outcome.   Thanks for the update!