realsenseai/librealsense #8123: Announcement: Introducing the 'Intel RealSense ID' facial authentication system

MartyG-RealSense, 2021-01-06:
Hi everyone,

Intel have introduced **Intel® RealSense™ ID**, an on-device solution that combines an active depth sensor with a specialized neural network designed to deliver secure, accurate and user-aware facial authentication.  Intel RealSense ID works with smart locks, access control, point-of-sale, ATMs, kiosks and more.

**Intel Newsroom press release**
https://newsroom.intel.com/news/introducing-intel-realsense-id-facial-authentication/#gs.pwpugd

[image]

solua100, 2021-01-07:
That's great.
Can you show me sdk for linux and android. Example code so simple.

MartyG-RealSense, 2021-01-07:
**GENERAL NOTE**:. I will only provide answers that are directly quotable from official Intel material in order to ensure accuracy of information.

**************

Hi @solua100  The link below leads to sample code for **RealSense ID Solution** in the C++, C# and C languages.

https://dev.intelrealsense.com/docs/sample-code-for-intel-realsense-id-solution

The **Tech Specs** section of the product information at https://www.intelrealsense.com/facial-authentication/ provides the following information about supported operating systems at the time of writing this:

– Windows
– Linux (Coming soon)
– Android (Coming soon)

The **Product Brief** PDF document provides the following information about an SDK:

Intel RealSense ID SDK is designed to plug into your user authentication systems.  Scalable, with configuration flexibility for edge, local host or server based solutions.

ShettyHarapanahalli, 2021-01-08:
Happy to see a splash-proof product from the real sense. 

By any chance can we deploy a neural network of our choice for a custom application
also, can we expect Open3d-ML library integration for this device?

MartyG-RealSense, 2021-01-08:
The **Intel Newsroom** press release states that RealSense ID uses a "specialized" and "dedicated" neural network.

The official Intel material currently available does not make reference to **Open3d-ML**.

closed 2021-01-09 by MartyG-RealSense