realsenseai/librealsense #8124: Dimensional Weight Software (DWS) Sample Code gives Compiler Error on Visual Studio 2019

MSari55, 2021-01-07:
When I try to compile DWS Sample Code as a Console application for Visual Studio 2019 (Windows 10, 64bit), I get an error "Unresolved external symbol".
https://www.intelrealsense.com/dimensional-weight-software/

I think I couldn't link the DWS_Sdk.dll to my project. 
(I added the folder containing DWS_SDK.dll to "Application->Properties->C/C++-General->Additional Include Directories" settings but still not working)

[image]

RealSenseSupport, 2021-01-11:
Hi,

Additional Include Directories should point the compiler to the SDK header (*.h) files, not the DLLs. Additional Libs directory should point the linker to the *.lib files...and the corresponding runtime DLLs need to just be int he same folder as your executable to resolve at run time.

Hope this helps,

Thanks

RealSenseSupport, 2021-01-20:
Hi,

Will you be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

MSari55, 2021-01-20:
[image]

Unfortunately, I am still getting compile errors.

I have defined the additional inclue and library paths where DWS_SDK dll files are loaded.
It's as if the system is looking for the DWS_sdk.cpp file.

When I install DWS SDK, I see files named DWS_SDK.h and DWSSDK.dll. (And other DLL and application files)

Could someone share me a visual studio project with .sln extension instead of .cpp code?

RealSenseSupport, 2021-01-20:
And you have DWSSDK.lib set as an additional dependency in the linker settings?

MSari55, 2021-01-20:
Yes, it worked after setting DWSSDK.dll to the additional dependency of linker.

Thank you.

closed 2021-01-20 by MSari55

nguyentrannhan, 2023-11-09:
I was unable to locate the DWS SDK download page. Could you please provide me with the link so I can download it?

MartyG-RealSense, 2023-11-09:
Hi @nguyentrannhan The Dimensional Weight Software (DWS) software, which only works with the RealSense L515 lidar depth camera model, was retired by Intel in 2021 and is no longer available, unfortunately.

An alternative free open-source code example for measuring the volume of a box is Intel's Python **box_dimensioner_multicam** project at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

A commercial box dimensioning product that uses RealSense technology is LIPS Corporation's **LIPSMetric Parcel Kiosk**

https://www.lips-hci.com/post/lipsmetric-parcel-kiosk-non-cubic-object-measurement

Another example of a RealSense-based commercial box dimensioning system is the one provided by MobileDemand.

https://www.ruggedtabletpc.com/blog/xDIM-software-launch