realsenseai/librealsense #8125: Installation failed

LyJoan, 2021-01-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.02.00| 
| Operating System & Version |  Linux (Ubuntu 18 | 
| Kernel Version (Linux Only)    | 5.3.0-53-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  C/C++/opencv/pcl/python                        | 
| Segment			|  Robot                        | 

### Issue Description
Yesterday I wanted to install RealSense SDK in my PC. I downloaded the latest version from github. When I run ./scripts/patch-realsense-ubuntu-lts.sh, I met a trouble which is
> "Replacing videobuf2-common  -
> cp: cannot stat '/home/cq/5.3.0-53-generic-videobuf2-common.ko': No such file or directory"

The log produced by running the script in the terminal is :

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> bc is already the newest version (1.07.1-2).
> build-essential is already the newest version (12.4ubuntu1).
> git is already the newest version (1:2.17.1-1ubuntu0.7).
> linux-headers-generic is already the newest version (4.15.0.129.116).
> The following package was automatically installed and is no longer required:
>   dkms
> Use 'sudo apt autoremove' to remove it.
> 0 upgraded, 0 newly installed, 0 to remove and 391 not upgraded.
> Package required libusb-1.0-0-dev:  - found
> Package required libssl-dev:  - found
> Create patches workspace in ubuntu-bionic-5 folder
> 
> Package required libelf-dev:  - found
> Package required elfutils:  - found
> Package required bison:  - found
> Package required flex:  - found
> Fetching Ubuntu LTS tag Ubuntu-hwe-5.3.0-53.47_18.04.1  to the local kernel sources folder
> The kernel has modified files:
> 	modified:   drivers/iio/accel/hid-sensor-accel-3d.c
> 	modified:   drivers/iio/gyro/hid-sensor-gyro-3d.c
> 	modified:   drivers/media/usb/uvc/uvc_ctrl.c
> 	modified:   drivers/media/usb/uvc/uvc_driver.c
> 	modified:   drivers/media/usb/uvc/uvcvideo.h
> 	modified:   drivers/media/v4l2-core/v4l2-ioctl.c
> 	modified:   include/uapi/linux/videodev2.h
> Proceeding will reset all local kernel changes. Press 'n' within 3 seconds to abort the operation
> Do you want to proceed? [Y/n]
> Resetting local changes in ubuntu-bionic-5 folder
>  HEAD is now at a70cc238a029 UBUNTU: Ubuntu-hwe-5.3.0-53.47~18.04.1
> Switching to LTS tag Ubuntu-hwe-5.3.0-53.47_18.04.1
> HEAD is now at a70cc238a029 UBUNTU: Ubuntu-hwe-5.3.0-53.47~18.04.1
> Applying patches for bionic-5 line
> Applying realsense-uvc patch
> patching file drivers/media/usb/uvc/uvc_driver.c
> Hunk #1 succeeded at 214 (offset -5 lines).
> patching file drivers/media/usb/uvc/uvcvideo.h
> patching file drivers/media/v4l2-core/v4l2-ioctl.c
> Hunk #1 succeeded at 1308 (offset 6 lines).
[... 206 lines cut ...]
>   CC      /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/v4l2-mem2mem.mod.o
>   CC      /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videobuf-dma-sg.mod.o
>   CC      /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videobuf-core.mod.o
>   CC      /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videodev.mod.o
>   CC      /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videobuf-vmalloc.mod.o
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/v4l2-dv-timings.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/v4l2-flash-led-class.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/tuner.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videodev.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/v4l2-common.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/v4l2-fwnode.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videobuf-vmalloc.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videobuf-dma-sg.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/v4l2-mem2mem.ko
>   LD [M]  /home/cq/librealsense/ubuntu-bionic-5/drivers/media/v4l2-core/videobuf-core.ko
> make: Leaving directory '/home/cq/librealsense/ubuntu-bionic-5'
> Patched kernels modules were created successfully
> 
> build_usbcore_modules=0
> Replacing videodev  -
> 	Applying the patched module ...  succeeded
> Replacing videobuf2-common  -
> cp: cannot stat '/home/cq/5.3.0-53-generic-videobuf2-common.ko': No such file or directory

Please tell me how to resolve these issues so any help would be greatly appreciated. Thanks!

ev-mp, 2021-01-07:
@LyJoan hello, for the kernel patching please use the recommended method -
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages
```sh
sudo apt install librealsense2-dkms
```

RealSenseSupport, 2021-01-14:
@LyJoan - Were you able to get through the instructions successfully?

RealSenseSupport, 2021-01-21:
@LyJoan - Were you able to get through the instructions successfully?

*If we don’t hear from you in 7 days, this issue will be closed.

eliasdc, 2021-01-28:
I'm having the same problem when building from source. During execution of the patches by default it does not compile videobuf2-common and videobuf2-v4l2 but it seems to be required for kernel >= 5. By specifying the `uvc_dbg` flag to the command I was able to build.

RealSenseSupport, 2021-01-28:
Thanks for the feedback.

TPEmist, 2021-03-04:
Same issue here too,

even with `sudo apt install librealsense2-dkms` still came up with the error 
`Replacing videobuf2-common  -
cp: cannot stat '/home/username/5.4.0-66-generic-videobuf2-common.ko': No such file or directory
`
OS : Ubuntu 18.04
Kernal Virsion : 5.4.0-66-generic

lyc-king, 2021-03-06:
Could you please tell more details about your way to solve the cp problem please? Thank you very much! @eliasdc

> I'm having the same problem when building from source. During execution of the patches by default it does not compile videobuf2-common and videobuf2-v4l2 but it seems to be required for kernel >= 5. By specifying the `uvc_dbg` flag to the command I was able to build.
> 
> If `sudo apt install librealsense2-dkms` should fix the problem as well I would suggest updating the installation guide so we know we can do either way and still use the latest master version.

lyc-king, 2021-03-06:
Now I know how to solve this and it works for me, sorry to bother you hahaha. @eliasdc

TPEmist, 2021-03-06:
Hi @lyc-king !
Cloud you kindly share the way to fix this. Thanks a lot.

> Now I know how to solve this and it works for me, sorry to bother you hahaha. @eliasdc

lyc-king, 2021-03-06:
> Hi @lyc-king !
> Cloud you kindly share the way to fix this. Thanks a lot.
> 
> > Now I know how to solve this and it works for me, sorry to bother you hahaha. @eliasdc

Sure, my pleasure. You can refer to #8513 which was created and closed by myself. There is detailed information in it.

RealSenseSupport, 2021-03-13:
Hi,

@LyJoan, will you still be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

seyoung-solteq, 2021-03-19:
I ran into the same error on Ubuntu 18.04 LTS.

## Cause of the problem
When I run `./scripts/patch-realsense-ubuntu-lts.sh`, it will make a directory `ubuntu-bionic-hwe-5.4` and fetch a tag `Ubuntu-hwe-5.4-5.4.0-67.75_18.04.1` from git://kernel.ubuntu.com/ubuntu/ubuntu-bionic.git.

In that tag, `ubuntu-bionic-hwe-5.4/drivers/media/common/videobuf2/videobuf2-common.ko` does not exist so this won't copy anything.

https://github.com/IntelRealSense/librealsense/blob/8fc9ad20da01e926ba610ce4ee1b058e61b86655/scripts/patch-realsense-ubuntu-lts.sh#L251

The content of the tag can be found here: https://kernel.ubuntu.com/git/ubuntu/ubuntu-bionic.git/tree/drivers/media/common/videobuf2?h=Ubuntu-hwe-5.4-5.4.0-67.75_18.04.1

[image]

-------------
## What else did I try?

I followed the following and building now. I will update once the build is over.

> @LyJoan hello, for the kernel patching please use the recommended method -
> https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages
> 
> sudo apt install librealsense2-dkms

seyoung-solteq, 2021-03-22:
> @LyJoan hello, for the kernel patching please use the recommended method -
> https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages
> 
> sudo apt install librealsense2-dkms

This didn't help.
`realsense-viewer` failed without any message.
When I ran `rs-capture` it failed with
```
librealsense Could not open OpenGL window, please check your graphic drivers or use the textual SDK tools
```

seyoung-solteq, 2021-03-22:
Rebooting the machine helped. :)

closed 2021-04-02 by RealSenseSupport

bigbellmercy, 2021-04-08:
Hello, I happened the same build error and report the workaround suggested by #8513:

### Workaround
In the `patch-realsense-ubuntu-lts.sh`, make the line 308 and 309 as comments by adding `#` at each beginning of line.

**we are not sure that this workaround does not make any problem later**. So I think this problem is not closed yet the root cause of this problem disappear.

Thanks.

### Environmemt
Ubuntu 18.04 x86 kernel v5.4.0-71-generic