realsenseai/librealsense #8126: 99-realsense-libusb.rules needs an update

pingplug, 2021-01-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Arch Linux  | 
| Kernel Version (Linux Only)    |  5.10.4                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.41.0                          | 

### Issue Description

```
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:3 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:4 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:5 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:6 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:7 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:8 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:9 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:10 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:11 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:12 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:13 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:14 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:15 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:16 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:17 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:18 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:19 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:20 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:21 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:22 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:23 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:24 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:25 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:26 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:27 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:28 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:29 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:30 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:31 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:32 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:33 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:34 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:35 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:38 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:39 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:40 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:41 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:44 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:45 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:46 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:48 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:50 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:52 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:54 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:56 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:58 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:60 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:62 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:64 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:66 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:68 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:70 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:72 Unknown group 'plugdev', ignoring
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:76 Invalid value "/bin/sh -c '(major=`uname -r | cut -d "." -f1` && minor=`uname -r | cut -d "." -f2` && (([ $major -eq 4 ] && [ $minor -ge 15 ]) || [ $major -ge 5 ])) && (enamefile=/sys/%p/name && [ `cat $enamefile` = "accel_3d" ]) && enfile=/sys/%p/buffer/enable && echo "COUNTER=0; while [ \$COUNTER -lt 20 ] && grep -q 0 $enfile; do sleep 0.01; COUNTER=\$((COUNTER+1)); done && echo 0 > $enfile" | at now'" for RUN (char 188: invalid substitution type), ignoring, but please fix it.
```

RealSenseSupport, 2021-01-19:
'plugdev' is pre-defined in Debian based Linux distributions.
https://wiki.debian.org/SystemGroups

You can add `plugdev` to you system. and add youself into the group.
`
# groupadd plugdev
# usermod -a -G plugdev username
`

pingplug, 2021-01-20:
realsense can not be mounted

Relying on `plugdev` is treated as a bug there and should be avoided.

https://github.com/Yubico/libu2f-host/issues/116

> You can add `plugdev` to you system. and add youself into the group.

still has an error there
```
systemd-udevd[253]: /etc/udev/rules.d/99-realsense-libusb.rules:76 Invalid value "/bin/sh -c '(major=`uname -r | cut -d "." -f1` && minor=`uname -r | cut -d "." -f2` && (([ $major -eq 4 ] && [ $minor -ge 15 ]) || [ $major -ge 5 ])) && (enamefile=/sys/%p/name && [ `cat $enamefile` = "accel_3d" ]) && enfile=/sys/%p/buffer/enable && echo "COUNTER=0; while [ \$COUNTER -lt 20 ] && grep -q 0 $enfile; do sleep 0.01; COUNTER=\$((COUNTER+1)); done && echo 0 > $enfile" | at now'" for RUN (char 188: invalid substitution type), ignoring, but please fix it.
```

RealSenseSupport, 2021-02-03:
Thank you for your feedback. we will continue to investigate this issue.

elflocoo, 2022-02-08:
Got the same issue, with:
Arch Linux
Kernel: 5.16.7-arch1-1
RealSense ROS v2.3.2
LibRealsense v2.50.0
2x RealSense D435

When launching two cameras, one does not start properly. When launching each of them individually, they work. Already tried it on several usb-ports and switched the cables, to exclude this issue. I also tried to replug the camera several times aswell. It is also not camera specific, because the camera, that won't load properly, changes (when pluging in, after another, its always the 2nd camera).

I also tried adding the plugdev group, but the same problem occurs, when launching the realsense cameras

Here the output of my launch process:

> Got the same issue, with:
Arch Linux
Kernel: 5.16.7-arch1-1
RealSense ROS v2.3.2
LibRealsense v2.50.0

When launching two cameras, one does not start properly. When launching each of them individually, they work. Already tried it on several ports and witched the cables, to exclude this issue. It is also not camera specific, because the camera, that won't load, changes.

Here the output of my launch process:

` roslaunch realsense2_camera rs_multiple_devices.launch 
... logging to /home/huf/.ros/log/e4e09bf6-88d6-11ec-a5bd-ac74b1a360a6/roslaunch-huf-11103.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://huf:34151/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/confidence_fps: 30
 * /camera1/realsense2_camera/confidence_height: 480
 * /camera1/realsense2_camera/confidence_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
[... 219 lines cut ...]
[ INFO] [1644323669.990171610]: Stereo Module was found.
[ INFO] [1644323669.993204384]: RGB Camera was found.
[ INFO] [1644323669.993233548]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1644323669.993247826]: num_filters: 0
[ INFO] [1644323669.993257190]: Setting Dynamic reconfig parameters.
 08/02 13:34:29,990 WARNING [140007065876032] (backend-v4l2.cpp:1387) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ WARN] [1644323670.012212828]: Param '/camera2/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1644323670.013422903]: Done Setting Dynamic reconfig parameters.
[ INFO] [1644323670.013724841]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1644323670.014006846]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1644323670.014018655]: setupPublishers...
[ INFO] [1644323670.014986825]: Expected frequency for depth = 30.00000
[ INFO] [1644323670.027115526]: Expected frequency for color = 30.00000
[ INFO] [1644323670.035151070]: setupStreams...
 08/02 13:34:30,060 WARNING [140091883488832] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 08/02 13:34:30,060 WARNING [140007091054144] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 08/02 13:34:30,060 WARNING [140091883488832] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 08/02 13:34:30,060 WARNING [140007091054144] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
[ INFO] [1644323670.142671272]: SELECTED BASE:Depth, 0
[ INFO] [1644323670.157390174]: RealSense Node Is Up!
[ WARN] [1644323670.212206200]: 
 08/02 13:34:31,156 ERROR [140006638634560] (ds5-options.cpp:88) Asic Temperature value is not valid!
 08/02 13:34:32,159 ERROR [140006638634560] (ds5-options.cpp:88) Asic Temperature value is not valid!

`

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense