realsenseai/librealsense #8128: Illegal CPU instruction!

yugg-github, 2021-01-08:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

MartyG-RealSense, 2021-01-08:
Hi @yugg-github   I hope that the link below will answer your question about this error mesage.

https://github.com/IntelRealSense/librealsense/issues/7080#issuecomment-671807292

The solution is to use the standalone **Intel.RealSense.Viewer.exe** version of the RealSense Viewer that can be downloaded from the **Assets** file list on the Releases page.  Once downloaded, this file should be launched with a double-leftclick from the folder location that it was downloaded to (not with the Viewer launch shortcut on the Windows desktop).

https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0

[image]

MartyG-RealSense, 2021-01-14:
Case closed due to correct answer provided and no further comment received.

closed 2021-01-14 by MartyG-RealSense