realsenseai/librealsense #8130: error while loading shared libraries: librealsense2-gl.so.2.34

fightmoney, 2021-01-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |T265 | 
| Firmware Version                   | 2.40 | 
| Operating System & Version |    Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  5.4.0  | 
| Platform                                 | PC |
| SDK Version                            |  2.4.0                         | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### I installed the SDK a mouth ago, it worked well !
But  there is something wrong with it ,so I use `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`  to uninstall it. And I find and delete anything about librealsense,  just like `sudo rm -r librealsense2*`.
Then I reinstall the SDK，follow the librealsense/doc/distribution_linux.md, and then `sudo apt update && sudo apt upgrade`
but `realsense-viewer: error while loading shared libraries: librealsense2-gl.so.2.34: cannot open shared object file :No such file or directory`

ev-mp, 2021-01-09:
@fightmoney ,  there is a discrepancy between the declared (and probably installed) 2.40 and the actual runtime SDK version 2.34.
 Apparently you have than one version of the SDK available, so you need to select the proper one to compile and run.
Use `ldd  exe_file_name` to list the actual SO files being loaded and their location on HD.

Also note that the uninstall routine you specified applies if the SDK was deployed with `sudo apt install ...`. When building from source code then the removal shall be done with `sudo make uninstall`.

fightmoney, 2021-01-11:
@ev-mp thanks for you reply, but there are some questions. I find librealsense location in my compute. So you mean I have to delete all of this?
[image]

ev-mp, 2021-01-11:
@fightmoney , it is not recommended to modify the content of system libs as it can break Ubuntu. You need to review those files and decide based on the installation you made and the intended usage.
In general there is nothing wrong with having multiple versions installed, however you need to be precise about the version you use in building/linking and running phases.

One method to avoid this complexity is to compile the SDK as static lib, but then again - it depends on the installation type and other dependencies that you might have.

fightmoney, 2021-01-12:
@ev-mp I am a new hand in linux, I think your advice maybe difficult for me. 
If I reinstall the version of 2.34.0 rather than 2.41.0, can it work?  How to install it?

closed 2026-10-23 by MartyG-RealSense