realsenseai/librealsense #8132: Working witj L515 and D435 in Phyton 3.8

lightov73, 2021-01-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 D435} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python }                          | 
| Segment			|  {others }                          | 

I try to install the phyton library for my two realsense camera's but I see that it supports only phyton 2 and I'm working with phyton 3.8 at the moment.
Also, is there a support for the LIDAR L515 ?
Ori

ev-mp, 2021-01-10:
@lightov73 hello,
> I try to install the python library for my two realsense camera's but I see that it supports only python 2 and I'm working with python 3.8 at the moment.

Standard python installers for Win/Ubuntu x86 are available for Pythons 2.7, 3.5,3.6 and 3.7. Support for Python 3.8 is in the backlog as well.

>Also, is there a support for the LIDAR L515 ?

Yes, L515 support was added with SDK release v2.34 (Apr 2020). 
See [Release Page](https://github.com/IntelRealSense/librealsense/releases) for details.

lightov73, 2021-01-12:
Thanks it is working now with my D435 but I didn't find any example that is working with the L515 although the project states that it is supposed to work with the L515. What do I miss ?

lightov73, 2021-01-20:
Hello - is there an answer?

RealSenseSupport, 2021-01-21:
Hi @lightov73 

I looked over some of the examples for python with regards to L515.  In order to get a few to work, I had to make modifications to the resolutions that were in some of the examples to work for the L515.  Some resolutions are not compatible across the L515 and the D400 cameras.  

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

Looking at the examples, I kept it pretty simple.  
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline - This sample I didn't have to make any modifications.

In the frame queue example, I needed to make a modification to the stream line from 640x360 to 640x480 and I was able to run the example
https://github.com/IntelRealSense/librealsense/blob/c7fbb7af032162f250c39322a86de9379df0bfd9/wrappers/python/examples/frame_queue_example.py#L33 
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

A good number of the examples were done a long time ago prior to L515 camera launch and yes the examples do work for the most part but may also need some tweaking in order to use functions/features/resolutions that are supported by L515.

Hope this helps.

lightov73, 2021-01-23:
Hi
    Thanks for the answer.
    I tried the above example nd they worked.
    But when I'm trying different ones I still got errors.
    When trying
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py
(changing the resolution as mentioned) I get

  File "C:\Users\משתמש\Google Drive\CO-EXIST (1)\המערכת והתהליכים\פיתוח
ומדגימים\DEV\Intel\examples\opencv_viewer_example.py", line 19, in <module>
    pipeline.start(config)

Its the same for other examples using also the color stream.

Ori

RealSenseSupport, 2021-02-01:
Hi,

Could you run the _rs-enumerate-devices_ SDK tool with your L515 connected and share the output with us?

Thanks

RealSenseSupport, 2021-02-15:
Hi,

Will you be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

lightov73, 2021-02-16:
Hi
Thabks
It took a whike but I manged to use the device with most of its
functionalities.
Ori

RealSenseSupport, 2021-02-19:
That's great. We'll close this issue then.

Should you need any additional assistance feel free to open another one at any time.

Thanks

closed 2021-02-19 by RealSenseSupport