realsenseai/librealsense #8136: How to use the smooth delta param in temporal filter?

7oud, 2021-01-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | 05.12.10 | 

### Issue Description
In my situation, I need to measure a relatively fixed range, about 1.8m to 2.2m, and the measurement errors (oscillations) mostly are no more than 25mm. So I used `temporal_filter` with `delta threshold` 20 to smooth oscillations within 20mm. Filter is not used in disparity domain, bcz of extra computation and a not large measurement range. 
In order to decrease the oscillations, I tried some methods, including a smaller depth units of 0.0001, however when using depth units 0.0001, delta 20 in temporal_filter needs to scaled by 10, which exceeds the max of delta 100. Does it mean that I cannot use the depth units of 0.0001 and delta of 20 in distance domain temporal_filter simultaneously. Can the problem be solved in disparity domain?
The other question is `How to improve measurement accuary in this range`, I tried methods including 
- A factor of 0.08
- High laser power
- High density preset
- Post-process of depth
- Smaller depth units

Is there any method else which may help to improve accuary in this range? What about self-calibration and tare-calibration on chip?

MartyG-RealSense, 2021-01-11:
Hi @7oud  At a range of 1.8 to 2.2 meters, the default depth unit scale of **0.001** should be fine.

Rather than using the High Density preset, **Medium Density** combined with a high **Laser Power** value may work better for you if accuracy is a priority, as this setting provides a balance between accuracy and fill-rate.  Please try these settings without any post-processing filters applied and see if it improves your results.

MartyG-RealSense, 2021-01-18:
Hi @7oud   Do you require further assistance with this case, please?   Thanks!

7oud, 2021-01-18:
@MartyG-RealSense Thanks for your help!

closed 2021-01-18 by 7oud

MartyG-RealSense, 2021-01-18:
You are very welcome @7oud - thanks very much for the update!