realsenseai/librealsense #8140: frame sync question in example

7oud, 2021-01-12:
In the example [Do Processing on a Background-Thread](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#do-processing-on-a-background-thread), if the data processing cost long time, will the main thread modify the `frame` which is being processed by child thread, or the newly arrived frame can not be enqueued?
```
rs2::pipeline pipe;
pipe.start();

const auto CAPACITY = 1;
rs2::frame_queue queue(CAPACITY);
std::thread t([&]() {
    while (true) {
        rs2::depth_frame frame;
        if (queue.poll_for_frame(&frame)) {
	        frame.get_data();
	        // Do processing on the frame, (cost long time)
        }
    }
});
t.detach();

while (true)
{
    auto frames = pipe.wait_for_frames();
    queue.enqueue(frames.get_depth_frame());
}
```

ev-mp, 2021-01-12:
The frame's lifespan  is controlled by the SDK with the frames memory being recycled. And while a `frame` is passed to the user app, e.g.. captured with `poll_for_frames`,  its content guaranteed until all references to it are released (frames are shared_ptr).
For each stream an internal memory queue is managed with default size of ~16 frames. Thus it is possible to acquire and keep up to that amount of frames in application before SDK reporting "No frames arrived".
In case more frames are required to be simultaneously available in the application , then you should use `frame.keep()` API (#6146).

7oud, 2021-01-13:
@ev-mp Thanks for your reply! In my psedo code, I use mutex to process frame manually, make the main thread to display image at 30FPS and the child thread may process image at lower FPS. The code is implemented by callback, Could it be achieved by `frame_queue` , `wait_for_frame` and `poll_for_frame`, if give some psedo code, that will be great.
```
mutex _mutex;
condition_variable _vc;
bool _frame_ready = false;

frame _color_frame;
frame _depth_frame;

void main()
{
    thread t(cbfn_process_frame);
    t.detach();

    auto cbfn_acquire_frame = [&](const frame& frame) {
        unique_lock<mutex> lck(_mutex);

        auto fs = frame.as<frameset>();
        _color_frame = fs.get_color_frame();
        _depth_frame = fs.get_depth_frame();

        // filter depth frame & convert color frame(yuv422) to yuv420
        _depth_frame = filter.process(_depth_frame);
        auto yuv420 = convert_frame_to_yuv420(color_frame);

        // unlock ASAP
        _frame_ready = true;
        _vc.notify_one();
        lock.unlock();

        send_image_to_display(yuv420);
    }

    pipeline.start(config, cbfn_acquire_frame);
    while (true)
        usleep(10000);
}

void cbfn_process_frame()
{
    while (true) {
        // if a new frame is acquired, then process
        unique_lock<mutex> lck(_mutex);
		_vc.wait(lck, []() -> bool { return _frame_ready; });

        // mat is a deep copy, not the same memory with frame
        cv::Mat color_mat = convert_frame_to_mat(_color_frame);
        cv::Mat depth_mat = convert_frame_to_mat(_depth_frame);

        // unlock ASAP
        lck.unlock();

        do_process_cost_long_time(color_mat, depth_mat);

        _frame_ready = false;
    }
}

```

ev-mp, 2021-01-13:
@7oud , I think there is a bug in your code - you define `_depth_frame` as global member, and then you modify it from different threads. So you need to modify the design and pass the frame from callback to the thread via `queue`.
There is no need in a mutex as the `queue` object will take care of it.

By __overriding the value__ in `_depth_frame = filter.process(_depth_frame);` you're __not changing the content of the frame__ (which is immutable by design), __but modifies the frame it points to__, so the thread that makes the heavy processing will be "surprised" and not in a good way...

7oud, 2021-01-13:
@ev-mp Actually, depth frame only be modified(filtered) in `acquire_callback`, and just be read in `fn_process_frame`. And filtered `_depth_frame` is converted to Mat type, which is shared as global variable. So I modify the psedo code in place.
The question is `How to use queue and wait/poll_for_frames` to rewrite the code. I don't know if not using mutex manually, how to release the global resource ASAP, which maintains the main thread's FPS

closed 2021-01-26 by 7oud