realsenseai/librealsense #8143: Realsense-Viewer depth frame representation variation with High accuracy preset.

07hokage, 2021-01-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| Segment			| Realsense Viewer   | 
| Version                          | v2.38.1 |

### Issue Description

In Realsense viewer, when i **activate stereo module** after setting **high accuracy preset,** they depth frame looks as attached in **pic 1**.  It looks **very sparse**. But after **activating also color module** **and deactivating** it again,  The stereo module now shows the depth frame as attached in **pic 2** which **seem to be the true representation**.  Is this a small bug, or does it create any issue if i load the high accuracy preset using sdk in my application?

**Pic1 -** before activating color module

[image]

**Pic2 -** after activating and deactivating color module

[image]

MartyG-RealSense, 2021-01-12:
Hi @07hokage   Image detail does tend to be sparser on the **High Accuracy** preset, so it is not a bug.   You may find that the **Medium Density** preset works better for you, as it provides a balance between accuracy and image "fill-rate".  

I was able to replicate the results of your tests.  

In the 3D point cloud mode of the RealSense Viewer program, activating RGB after the Stereo Module adds a **Color** menu option to the **Texture** option that is at the top of the RealSense window when in 3D mode.  Having RGB enabled in 3D mode makes it possible to map between color and depth points, which can enhance the detail on the image.

Below is shown an example of selecting Color as the texture to map RGB to depth.

[image]

07hokage, 2021-01-13:
@MartyG-RealSense thank you for the clarification. So this is mostly related to visualization only right?   i'm trying to use the high accuracy preset for robot obstacle detection. ( as suggested in documentation ). using realsense sdk  Does enabling both depth and color streams, affect the quality of my depth data any way?

MartyG-RealSense, 2021-01-13:
Aligning depth and color helps to distinguish foreground pixels from background pixels.  It is also useful for removing (segmenting) color background detail, as demonstrated by the SDK example program **rs-align-advanced**.  The example shows how color detail can be excluded from the image when it is mapped to depth pixels and a maximum depth distance (depth clamp) is applied.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align-advanced

07hokage, 2021-01-13:
Thanks @MartyG-RealSense ! That certainly helps. Closing the issue now

closed 2021-01-13 by 07hokage