realsenseai/librealsense #8147: d435 instead of d435i

Moaaz997, 2021-01-13:
* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | RealSense Viewer  | 
| Operating System & Version |   ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140                                      | 
| Platform                                 | NVIDIA Jetson nano  |
| SDK Version                            |                             | 
| Segment			|  {Robot }                          | 

### Issue Description
 My realsense viewer shows that my camera version is D435 instead of (D435i which is the real camera model), what could be the reason behind this?

MartyG-RealSense, 2021-01-13:
Hi @Moaaz997   If you have a D435i camera and the camera name is show as D435 in the Viewer, it suggests that the IMU component inside the camera is not being detected.

I would recommend updating your camera's firmware driver to see whether doing so corrects the problem.  You can do this by left-clicking on the **More** option near the top of the options side-panel of the RealSense Viewer and selecting **Install Recommended Firmware** from its menu.

[image]

Moaaz997, 2021-01-13:
I have just installed the recommended firmware and nothing changed

MartyG-RealSense, 2021-01-13:
Thank you very much for performing the firmware update.

-  Does the camera remain identified as D435 instead of D435i if you reset the camera by selecting **Hardware Reset** from the same menu?

-  Could you tell me which RealSense SDK version you are using, please?

Moaaz997, 2021-01-19:
- I tried the hardware reset and it didn't work
- the installed SDK version is v2.31.0.
- this SDK version gives me the error:  Multiple realsense udev-rules were found!

MartyG-RealSense, 2021-01-19:
Multiple udev rules can indicate that librealsense has been installed on separate occasions on the same computer with both the dkms package and the source code methods of installation.  It can be corrected by deleting the **/etc** rule and not the **/lib** rule, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

The **/lib** rule is not deleted because doing so can potentially break linkages to Debian packages.

Moaaz997, 2021-01-21:
I face the following error, and when i execute the commands the file that i just deleted are restored.

`Timestamp: 1611249640151.504395
Severity: Warn
Description: RealSense UDEV-Rules file:
 /lib/udev/rules.d/60-librealsense2-udev-rules.rules
 is not up-to date!
UDEV-Rules permissions configuration 
 for RealSense devices.

Missing/outdated UDEV-Rules will cause 'Permissions Denied' errors
unless the application is running under 'sudo' (not recommended)
In case of Debians use: 
sudo apt-get upgrade/install librealsense2-udev-rules
To manually install UDEV-Rules in terminal run:
$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger`

MartyG-RealSense, 2021-01-21:
If you only wanted to update the rule related to the Debian package (**/lib**) then I would speculate that you should just run the update command for a package-based installation:

**sudo apt-get upgrade/install librealsense2-udev-rules**

Since you have deleted the **/etc** rule, do not run the instruction for installing **/etc**, which is:

**sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger`**

closed 2021-02-05 by MartyG-RealSense