realsenseai/librealsense #8148: depth infomation is lost after aligning depth to color

w-phenix, 2021-01-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 5.4.0-60-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.41.0 }                          | 
| Language                            |  python }                          | 
| Segment			|  others | 

### Issue Description
when I run [align-depth2color.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py), I found the showed pictures like this:
[image]

and when I debug, I found all the elements of depth matrix are zero:
[image]

so what can I do to get the aligned depth without losing infomation?

MartyG-RealSense, 2021-01-13:
Hi @w-phenix   The example program **align-depth2color.py** produced a similar depth image in a past case.

https://github.com/IntelRealSense/librealsense/issues/2501

As advised in the case, could you try changing the line **clipping_distance_in_meters = 1 #1 meter** to use a value higher than '1' (meter).  A good test value would be '4'.

w-phenix, 2021-01-14:
> Hi @w-phenix The example program **align-depth2color.py** produced a similar depth image output and zeroed depth values in a past case.
> 
> #2501
> 
> As advised in the case, could you try changing the line **clipping_distance_in_meters = 1** to use a value higher than '1' (meter). A good test value would be '4'.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L34

It doesn't seem to work. I tried from 2 to 4, even a bigger number 10, but the depth data are still zero and the color picture shows grey all the time. Maybe I should check if my camera is broken?

MartyG-RealSense, 2021-01-14:
As you stated that you can receive depth successfully with a non-aligned image, that would indicate that your camera is not broken.

Are you able to receive a color image too when not performing alignment, please?

w-phenix, 2021-01-14:
> As you stated that you can receive depth successfully with a non-aligned image, that would indicate that your camera is not broken.
> 
> Are you able to receive a color image too when not performing alignment, please?

Yes, I can receive  both non-aligned depth image and non-aligned color image, like this:
[image]

The code I used is [opencv_viewer_example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py)

MartyG-RealSense, 2021-01-14:
I researched your problem further but could not find an obvious reason why **align-depth2color.py** is not working for you.  Does the program close down if you press the Esc or Q key?  This would indicate that the program is at least responsive and not frozen.

w-phenix, 2021-01-15:
> I researched your problem further but could not find an obvious reason why **align-depth2color.py** is not working for you. Does the program close down if you press the Esc or Q key? This would indicate that the program is at least responsive and not frozen.

Yes, I can quit normally by pressing the Esc or Q key. Futhermore, I can receive aligned color image (depth data are still 0) after commenting out this line `bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)` and modifying this line to `images = np.hstack((color_image, depth_colormap))`
The output is as below:
[image]

MartyG-RealSense, 2021-01-15:
It looks as though the logic of this line:

**bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0)**

is basically saying to set to a grey color the pixels whose depth either exceeds the defined clipping distance, or is equal to or less than zero.

Since the left-hand side of the image in your original comment was colored completely grey, that would suggest to me that the program interpreted the depth as either being equal / less than zero (i.e no depth detected) or that all the measured depth values were outside of the defined clipping distance.  Indeed, going back to the original images in this discussion, all of the depth coordinates registered as zero, which may account for the entire left-hand side being colored grey.

It may be worth using the **Dynamic Calibrator** tool on your camera (which calibrates both depth and RGB) to check whether the problems when aligning are due to a mis-calibration between the sensors.

w-phenix, 2021-01-16:
@MartyG-RealSense I studied the issue you suggested, but I still can't resolve my problem. They got the alignment order wrong (process the alignment before starting the pipeline) while I didn't.
By the way, just in case, I tried to put the alignment code before creating a pipeline or starting streaming, they all get the same result as the original order.

MartyG-RealSense, 2021-01-16:
I carried out further careful research into your problem but could not find any other leads that might fix your problem with **align-depth2color.py** unfortunately.

The logical next step would be to try the alignment code from Intel's **distance_to_object** Python tutorial.

https://github.com/IntelRealSense/librealsense/issues/2560

w-phenix, 2021-01-17:
@MartyG-RealSense Yes, I tried **distance_to_object**, but encountered the same problem. So I think  the code is no problem, but the camera is something wrong.

MartyG-RealSense, 2021-01-17:
Thanks very much.  Another quick way to test the ability of the camera to align depth and color would be to map RGB to a depth point-cloud in the 3D mode of the RealSense Viewer.  The method for doing so is described in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360051738754/comments/360015204353

If the depth and RGB still do not align then you can perform a quick calibration of the camera by going to the **More** option near the top of the Viewer's options side-panel and selecting the **On-Chip Calibration** option from its menu.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

w-phenix, 2021-01-17:
@MartyG-RealSense I use the RealSense Viewer to observe color frame and depth frame, they shows as belows:
[image]
[image]
[image]

and I also tried **On-Chip Calibration**. However, the alignment code still doesn't work after I calibrate:
[image]
I tried both **intrinsic and extrinsic** (**speed fast** and **OCC All**)

MartyG-RealSense, 2021-01-17:
The alignment in the 3D mode of the Viewer also seems broken.

If On-Chip Calibration did not improve results then a more robust calibration of the depth and RGB sensors can be performed using the **Dynamic Calibrator** software tool.  Page 14 onwards of the tool's user guide instructs how to install it on Linux.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

w-phenix, 2021-01-18:
@MartyG-RealSense I installed the **Dynamic Calibrator** software tool as written in the user guide. Everything is good until I tried to calibrate the camera; The calibration unexpectedly failed to start:
[image]

MartyG-RealSense, 2021-01-18:
Are you using the camera in a USB 3 port please?  The tool was created before USB 2 support was introduced to the RealSense SDK and lists USB 3.1 in its Hardware Requirements documentation.

This error has also been reported to occur on a Windows computer with the RealSense UWP Driver version .**6.11.160.21** installed, though you are using Ubuntu and so that would not apply in this case.

w-phenix, 2021-01-19:
@MartyG-RealSense Can every camera eliminate the blue square in the middle? I tried for a long time but I couldn't completely eliminate it. After eliminating a part, the part that was eliminated before moving the camera will turn back to blue again. I don’t know if my method of elimination is wrong, or there are other things I didn’t notice...
[image]
[image]

MartyG-RealSense, 2021-01-19:
I have performed this procedure recently with a D435i myself, so it is certainly possible to eliminate all of the blue area.  It can be a little awkward though.

You should try to keep the camera at relatively the same distance throughout, and move it left, right, up and down until all of the blue area is removed.  For myself personally, the technique that worked best was following a regular movement pattern of the camera: up-down-left-right, up-down-left-right etc and repeat until all of the blue was removed. 

I also highly recommend having a high timeout value set so that you are not repeatedly returned to the start of the process if it is not completed within the permitted time.

MartyG-RealSense, 2021-01-26:
Hi @w-phenix   Do you require further assistance with this case, please?   Thanks!

w-phenix, 2021-01-30:
@MartyG-RealSense The camera still cann't perform the alignment, so our lab bought a new one. I will close this issue and thank you for your help!

closed 2021-01-30 by w-phenix