realsenseai/librealsense #8149: Error

AkisouMa-Young, 2021-01-13:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

MartyG-RealSense, 2021-01-13:
Hi @AkisouMa-Young  I hope that the link below will answer your question about this error mesage.

https://github.com/IntelRealSense/librealsense/issues/7080#issuecomment-671807292

The solution is to use the standalone **Intel.RealSense.Viewer.exe** version of the RealSense Viewer that can be downloaded from the **Assets** file list on the Releases page. Once downloaded, this file should be launched with a double-leftclick from the folder location that it was downloaded to (not with the Viewer launch shortcut on the Windows desktop).

https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0

[image]

MartyG-RealSense, 2021-01-19:
Case closed due to correct answer provided and no further comments received.

closed 2021-01-19 by MartyG-RealSense