realsenseai/librealsense #8152: Intel Realsense T265 camera not working on demo example (robot_devkit)

dbgarasiya, 2021-01-13:
----------------------------------------------------------------------------------------------------

 Camera Model                       | {T265 } | 
| Firmware Version                   | {0.2.0.951} | 
| Operating System & Version |   {Ubuntu 18.04.5 LTS}  | 
| Kernel Version (Linux Only)    |  (Linux 5.4.0-42-generic)    | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  {2.41.0-0}                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

I am working on projects with two intel camera ( D435i and T265)
after installing robot_devkit  when i play demo for both camera

D435i camera works but T265 camera not working , it gives error that device is not connected ( File name :T265_Error.txt 
[T265_Error.txt](https://github.com/IntelRealSense/librealsense/files/5808435/T265_Error.txt)

which gives in details.

my realsense-viewer command works for both camera ,

dbgarasiya, 2021-01-20:
**@MartyG-RealSense** , please suggest any solution

RealSenseSupport, 2021-01-20:
Hi @dbgarasiya 

Have you given the command outlined here a try:  https://github.com/IntelRealSense/realsense-ros/tree/eloquent#using-t265

The two cameras should work together as you pointed out they function properly separately in realsense-viewer.

How about if you only have the T265 connected and running ros command, does that work properly as well?

dbgarasiya, 2021-01-22:
Thanks for replying **@RealSenseSupport** 

I have followed link given by you
where  I run below command

ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_pose:=true -p device_type:=t265 -p fisheye_width:=848 -p fisheye_height:=800

i found that "**Package 'realsense2_camera' not found** "

when i try to install from your link i got error below point

git clone https://github.com/IntelRealSense/realsense-ros.git -b dashing

Error is "**fatal: Remote branch dashing not found in upstream origin** "

I have used dashing ROS2 version in my configuration

Best Regadrs

closed 2026-10-23 by MartyG-RealSense