realsenseai/librealsense #8155: Getting segmentation fault (core dumped) after code exits

KevinKenntec, 2021-01-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.10.0 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  18.04.5   | 
| Platform                                 | PC  |
| SDK Version                            |  2 | 
| Language                            |  C++   | 
| Segment			| Others  | 

### Issue Description
I get a segmentation fault at the absolute end of any code I currently run, past the "return 0" point (I placed a cout statement just before the return and it prints out that cout statement AND THEN gives a segmentation fault). All this code does is literally grab both a colour frame and a depth-frame (with a colour map), make them into Mat variables and then write them as jpg files via OpenCV. This issue, however, never appears if I run this code WITH a debugger like GDB.

This is all with a completely fresh install of Ubuntu 18.04.5 and a fresh install of OpenCV and also the Realsense libraries via: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

I also get this with the Intel Realsense-Viewer as well which I have raised here as Issue 8154: https://github.com/IntelRealSense/librealsense/issues/8154

Often after the camera has a segmentation fault, the camera's depth stream often goes completely back and either needs to be replugged or hardware reset to work again.

I have attached the main code I used as a text file rather than a cpp file.
[test.txt](https://github.com/IntelRealSense/librealsense/files/5811505/test.txt)

MartyG-RealSense, 2021-01-14:
Hi @KevinKenntec   As you are also experiencing the segmentation issue in the RealSense Viewer in case https://github.com/IntelRealSense/librealsense/issues/8154, I recommend keeping this case open but pausing it and focusing on https://github.com/IntelRealSense/librealsense/issues/8154 first in order to avoid duplicating the investigation.  Does that sound okay to you, please?

KevinKenntec, 2021-01-14:
Yes that would be fine.

MartyG-RealSense, 2021-01-14:
Thanks very much @KevinKenntec

MartyG-RealSense, 2021-02-14:
Case closed due to related case https://github.com/IntelRealSense/librealsense/issues/8154 being answered.

closed 2021-02-14 by MartyG-RealSense