realsenseai/librealsense #8160: T265 C# : Frame didn't arrive within 5000 problem

9and3, 2021-01-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |   Win 10  | 
| Language                            |  C#                          | 

### Issue Description
Hello! I am trying to retrieve pose data from the T265 tracking camera with C# wrappers. 
The pose is streamed for a while and than I got this error:

`Unhandled Exception: System.Exception: Frame didn't arrive within 5000 ---> System.Runtime.InteropServices.ExternalException: rs2_pipeline_wait_for_frames(pipe:00000232D8DF2340)
   --- End of inner exception stack trace ---
   at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged(IntPtr pNativeData)
   at System.StubHelpers.MngdRefCustomMarshaler.ConvertContentsToManaged(IntPtr pMarshalState, Object& pManagedHome, IntPtr pNativeHome)
   at Intel.RealSense.NativeMethods.rs2_pipeline_wait_for_frames(IntPtr pipe, UInt32 timeout_ms, Object& error)
   at Intel.RealSense.Pipeline.WaitForFrames(UInt32 timeout_ms)`

And here's my code:

`//Declare Realsense pipeline, encapsulating the actual device and sensors
            var pipe = new Pipeline();
            // Create a configuration for configuring the pipeline with a non default profile
            var cfg = new Config();
            // Add pose stream
            cfg.EnableStream(Stream.Pose, Format.SixDOF);
            // Start pipeline with chosen configuration
            pipe.Start(cfg);

            while (true)
            {
                using (var frames = pipe.WaitForFrames())
                using (var f = frames.FirstOrDefault(Stream.Pose, Format.SixDOF))
                {
                    // Cast the frame to pose_frame and get its data
                    var pose_data = f.As<PoseFrame>().PoseData;
                    // Retrieve data from pose
                    var trans = pose_data.translation;
                    Console.WriteLine("T pose: " + (double)trans.x + "," + (double)trans.y + "," + (double)trans.z);
                    f.Dispose();
                    frames.Dispose();
                }
            }`

I just would like to say that I tried the same code to run the tracking on C++, the same computer, and port USB and it runs perfectly. 

Any idea? Thanks in advance!

RealSenseSupport, 2021-01-19:
Hi,

We were able to run this code successfully using the latest SDK release, but when (experimentally) removing the using and Dispose, it then exhibits the behavior you're seeing, so a guess would be that somehow those frame objects are not actually being released on your end. The CLR maybe?

Hope this helps.

Thanks

9and3, 2021-01-22:
Hello,

thank your for your answer. I did actually tried on different machines and different NET versions, we are all still reading the same error. Isn't there a workaround to the using/Dispose formula? Maybe a different way of formulate the while loop? Thanks

RealSenseSupport, 2021-01-27:
Unfortunately no. See Note here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp#hello-world

Thanks

Dalinaffeti, 2021-01-29:
i had the same issue 
i used threading to start streaming from the camera 
are u on WinForms or WPF ?

9and3, 2021-01-31:
I believe I am in WPF. What would be the difference @Dalinaffeti I am interested? So the whole streaming loop would be done in a parallel thread? I will try out this, thanks!

Dalinaffeti, 2021-02-01:
exactly like that it is about parallel threading. the difference is a litte syntax that's it. it is easier to follow the realsense c# wrapper with the WPF. For WinForms there is nothing about that but that shouldn't be a problem i guess. BUT there is something that i want to confirm is the streaming slower with c#  or not

petersengm, 2022-09-08:
I'm having the same issue.  I tried running the rs_depth.c application and that also is throwing a timeout error.  I don't think this is a C# problem but related to the underlying native code.

9and3, 2022-09-12:
I agree. Personally I went back to the C++ API and did my own C# wrap just to call the functions I strictly needed for my applications. Didn't have a problem with that with buffers or anything else.

haldaazhim, 2023-06-26:
I had the same issue when I ran the camera on raspberry pi 4b, sometimes it doesnt freeze but sometimes it freeze. I dont surely know how to fix it and what causing the freeze

wartih, 2023-07-14:
> I had the same issue when I ran the camera on raspberry pi 4b, sometimes it doesnt freeze but sometimes it freeze. I dont surely know how to fix it and what causing the freeze↳

same problem :(

zhengshuyu1234, 2023-08-01:
I have the same problem when I  ran the t265 camera on  my raspberry pi 4b.

ha-ves, 2024-06-12:
You must dispose all frame(set)s associated with the pipeline before getting new one (unless you use the `.Keep()`).
GC is an expensive process, hence the wrapper insists on disposal as soon as it is not used directly.

for example, instead of just:
```c#
// This will cause blocking for the next call because the wrapper expect to reuse the same object
// You need to dispose each of them
using var frm = align.Process(p.WaitForFrames()).AsFrameSet();
```
you have to do this (for current scope):
```c#
using var f1 = p.WaitForFrames();
using var f2 = align.Process(f1);
using var frm = f2.AsFrameSet();
```
or (for a more global scope):
```c#
CamFrameSet = p.WaitForFrames();
Aligned = align.Process(CamFrameSet).DisposeWith(CamFrameSet).AsFrameSet().DisposeWith(CamFrameSet);
```
so if the source frame gets disposed, the derivatives gets disposed. Unless you choose to use `.Keep()`.

in @9and3's case, you must dispose this object:
```c#
f.As<PoseFrame>()
```
which will be:
```c#
// Cast the frame to pose_frame and get its data
var pose_data = f.As<PoseFrame>().DisposeWith(f).PoseData;
```
or you have to remove it from the pipeline's framepool
```c#
// Cast the frame to pose_frame and get its data
var tmp = f.As<PoseFrame>();
// put the frame out of the reuse cycle for the pipeline
tmp.Keep();
var pose_data = tmp.PoseData;
```
after this point, you're dependent on GC to dispose of `tmp` once it goes out of scope, which is expensive and not reliable which may cause memory leak and performance degradation.

closed 2026-10-23 by MartyG-RealSense