realsenseai/librealsense #8161: setting visual preset for l515 via python

eyebies, 2021-01-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L500} | 
| Firmware Version                   |  | 
| Operating System & Version |   {Ubuntu 18}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  NVIDIA Jetson.  |
| SDK Version                            |  { l 2.41.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
I am trying to set the RS2_L500_VISUAL_PRESET_SHORT_RANGE rs2_l500_visual_preset_short_range option for l515 from the python api. Just wondering it this is similar to the d400 setting . 
depth_sensor.set_option(rs.option.visual_preset, 4)
thanks !

RealSenseSupport, 2021-01-14:
The short answer is no. Visual Presets are a set of state that is hardware specific....and D400 uses different technology (stereo) than L500 (lidar)

Thanks

eyebies, 2021-01-15:
Thanks for coming  back to me! Would be possible  a snippet to set it from python? Thanks!

eyebies, 2021-01-18:
Would this work ?

presets = [rs.l500_visual_preset.custom,
           rs.l500_visual_preset.no_ambient_light,
           rs.l500_visual_preset.low_ambient_light,
           rs.l500_visual_preset.max_range,
           rs.l500_visual_preset.short_range]
preset =  rs.l500_visual_preset.short_range
depth_sensor.set_option(rs.option.visual_preset, int(preset))

closed 2021-01-18 by eyebies