realsenseai/librealsense #8163: ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)

monacv, 2021-01-14:
```
$ pip install pyrealsense2
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

$ pip --version
pip 20.0.2 from /home/mona/venv/vision/lib/python3.8/site-packages/pip (python 3.8)
$ python
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

```

tqtanh, 2021-01-15:
Hi, 

Based on the pre-build library packages for pip install on this site
https://pypi.org/project/pyrealsense2/#files
Currently the pyrealsense2 is supported to install by pip include python 2.7, 3.5, 3.6, and 3.7
[image]

You can downgrade your python to 3.7 or clone the library resporitory and rebuilt it with the option "BUILD_PYTHON_BINDINGS". See more at: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python.

Cheers

MartyG-RealSense, 2021-01-15:
Thanks so much @tqtanh  for your advice to @monacv 

Hi @monacv   Please let me know whether the advice above solves your problem or whether you require further assistance.   Thanks!

monacv, 2021-01-15:
I ended up installing from source so I don't need further help. Thank you.

closed 2021-01-15 by monacv

MartyG-RealSense, 2021-01-15:
Great to hear that you succeeded - thanks very much for the update!