realsenseai/librealsense #8164: ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2' when $ python setup.py bdist_wheel

monacv, 2021-01-14:
```
[10867:8660 0:2024] 02:34:57 Thu Jan 14 [mona@goku:pts/1 +1] ~/research/code/librealsense/wrappers/python
$ python setup.py bdist_wheel
Traceback (most recent call last):
  File "setup.py", line 5, in <module>
    from pyrealsense2._version import __version__
  File "/home/mona/research/code/librealsense/wrappers/python/pyrealsense2/__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'

```

I have:

```
$ python
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
$ pip --version
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)

$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

```

MartyG-RealSense, 2021-01-15:
Hi @monacv   I note that the first command that you input in the above logs is **python setup.py bdist_wheel**.  This indicates that you attempted to install the Python wrapper using the **pip** installation method in the Python wrapper documentation.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md

As mentioned in your other wrapper installation cases https://github.com/IntelRealSense/librealsense/issues/8163 and https://github.com/IntelRealSense/librealsense/issues/8167 a pip installation will not be successful with Python 3,8 or on a computing device with an Arm processor.

monacv, 2021-01-15:
I'm all set since I decided to install from source

closed 2021-01-15 by monacv