realsenseai/librealsense #8167: ModuleNotFoundError: No module named 'pyrealsense2' after successful $ sudo make install

monacv, 2021-01-14:
After `sudo make install`, still I cannot import it:
```
$ python estimate_plane.py 
Traceback (most recent call last):
  File "estimate_plane.py", line 9, in <module>
    import pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2'
```

```

$ ls /usr/local/lib/
total 607M
drwxrwsr-x  3 root staff 4.0K Jul 31 12:27 python3.8
-rwxr-xr-x  1 root root  4.7M Jan  5 21:42 libprotobuf-lite.so.17.0.0
lrwxrwxrwx  1 root root    26 Jan  5 21:42 libprotobuf-lite.so.17 -> libprotobuf-lite.so.17.0.0
lrwxrwxrwx  1 root root    26 Jan  5 21:42 libprotobuf-lite.so -> libprotobuf-lite.so.17.0.0
-rwxr-xr-x  1 root root  1011 Jan  5 21:42 libprotobuf-lite.la
-rwxr-xr-x  1 root root   42M Jan  5 21:42 libprotobuf.so.17.0.0
lrwxrwxrwx  1 root root    21 Jan  5 21:42 libprotobuf.so.17 -> libprotobuf.so.17.0.0
lrwxrwxrwx  1 root root    21 Jan  5 21:42 libprotobuf.so -> libprotobuf.so.17.0.0
-rwxr-xr-x  1 root root   976 Jan  5 21:42 libprotobuf.la
-rwxr-xr-x  1 root root   59M Jan  5 21:42 libprotoc.so.17.0.0
lrwxrwxrwx  1 root root    19 Jan  5 21:42 libprotoc.so.17 -> libprotoc.so.17.0.0
lrwxrwxrwx  1 root root    19 Jan  5 21:42 libprotoc.so -> libprotoc.so.17.0.0
-rwxr-xr-x  1 root root   992 Jan  5 21:42 libprotoc.la
-rw-r--r--  1 root root   12M Jan  5 21:42 libprotobuf-lite.a
-rw-r--r--  1 root root  109M Jan  5 21:42 libprotobuf.a
-rw-r--r--  1 root root  172M Jan  5 21:42 libprotoc.a
drwxr-xr-x 12 root root  4.0K Jan  8 20:10 ..
drwxrwsr-x  4 root staff 4.0K Jan 14 14:37 python2.7
-rw-r--r--  1 root root  1.7M Jan 14 15:04 libglfw3.a
-rw-r--r--  1 root root   13M Jan 14 15:04 libfw.a
-rw-r--r--  1 root root   12M Jan 14 15:04 librealsense-file.a
-rw-r--r--  1 root root  165M Jan 14 15:06 librealsense2.so.2.41.0
-rw-r--r--  1 root root   20M Jan 14 15:07 librealsense2-gl.so.2.41.0
lrwxrwxrwx  1 root root    23 Jan 14 15:12 librealsense2.so.2.41 -> librealsense2.so.2.41.0
lrwxrwxrwx  1 root root    21 Jan 14 15:12 librealsense2.so -> librealsense2.so.2.41
drwxr-xr-x  2 root root  4.0K Jan 14 15:12 pkgconfig
lrwxrwxrwx  1 root root    26 Jan 14 15:12 librealsense2-gl.so.2.41 -> librealsense2-gl.so.2.41.0
lrwxrwxrwx  1 root root    24 Jan 14 15:12 librealsense2-gl.so -> librealsense2-gl.so.2.41
drwxr-xr-x  6 root root  4.0K Jan 14 15:12 cmake
drwxr-xr-x  6 root root  4.0K Jan 14 15:12 .
```

and I have:
```
export PYREALSENSE2_PATH="/usr/local/lib:$PYREALSENSE2_PATH"
export PYTHONPATH="/usr/local/lib:$PYTHONPATH"
```

I also have:
```
$ python
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
$ pip --version
pip 20.0.2 from /home/mona/venv/vision/lib/python3.8/site-packages/pip (python 3.8)
$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

```

MartyG-RealSense, 2021-01-15:
Hi @monacv   Could you tell me which computer or computing device you are using please?  Bear in mind that:

1.  pip install cannot be used with pyrealsense2 for Python 3.8 and the Python wrapper therefore has to be installed from source code for 3.8, as discussed in https://github.com/IntelRealSense/librealsense/issues/8163

2.  If you are using a device with an **Arm** processor such as Raspberry Pi or Nvidia Jetson, you will still not be able to use pip install if you use Python versions earlier than 3.8, such as 3.6 or 3.7.  This is because the pip pyrealsense2 packages are only compatible with computers with **x86 or x64** processors and will not work with Arm processors.

Raspberry Pi and Nvidia Jetson have their own methods of building the Python wrapper.  If you have a Pi or Jetson then I will direct you to the appropriate installation information.

monacv, 2021-01-15:
actually I installed again without doing the sudo make install and working all the steps until end of make and that worked

closed 2021-01-15 by monacv