realsenseai/librealsense #8169: Install Dynamic Calibration on Raspberry Pi 4B

dingbaiyin, 2021-01-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D430 | 
| Firmware Version                   | 5.12.6 | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  {2.36.0}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
How to install Dynamic Calibration under Raspberry Pi 4B Ubuntu system,Is there a reference document

dingbaiyin, 2021-01-15:
Is there an Android phone dynamic calibration apk?

MartyG-RealSense, 2021-01-15:
Hi @dingbaiyin   Instructions for installing thr Dynamic Calibration tool on Linux can be found in the link below.

https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide#section-3-installation

You can use an Android device with the Dynamic Calibrator tool for the purposes of displaying a digital calibration target on the screen instead of printing off a target onto paper.

dingbaiyin, 2021-01-16:
An error occurred when I configured the installation
“N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repositor y 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease'  doesn't support architecture 'armhf'”
[image]
After I upgraded my pi4b system from 32bit to 64bit, it still cannot be installed normally. Is there a successful case?

dingbaiyin, 2021-01-16:
[image]

MartyG-RealSense, 2021-01-16:
The documentation for the Dynamic Calibrator states that although Arm is supported, Arm testing has been limited to Nvidia Jetson devices.

[image]

May I confirm please which Ubuntu version you are using?  If it is **16.04** then the Debian package installation instruction is:

echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list

If **18.04** is being used then the PDF version of the instructions provides the instruction below for **bionic** instead of xenial instead:

echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main' | sudo tee /etc/apt/sources.list.d/realsense-public.list

***************

It may also be worth checking whether your calibration needs can be met by using **On-Chip Calibration** instead.  This can be accessed from the menu of the **More** option near the top of the side-panel of the **RealSense Viewer** program.

[image]

dingbaiyin, 2021-01-18:
1.The Raspberry Pi 4 system version is as shown below
[image]
I referred to this datasheet when I installed the dynamic calibration software on the Raspberry Pi, but an error was reported, and the error message was replied last time
I have successfully installed RealSense Viewer 2.36, and On-Chip Calibration can also be used. But the dynamic calibration effect is better than the On-Chip Calibration method. So I want to implement dynamic calibration function on Raspberry Pi 4B.

2.If you replied to me last time, I learned that Android phones can only replace the calibration target with the phone but cannot act as a dynamic calibration host.
I would like to know if I want to use an Android phone as a dynamic calibration phone, is there such a possibility from the hardware point of view, if it is feasible, is there any reference method, thank you very much for your reply

MartyG-RealSense, 2021-01-18:
Linux can be installed on Android.  Details of this can be found by googling for **install Linux on android**, and an example of such an installation guide can be found in the link below.

https://www.lifewire.com/run-linux-on-android-4586926

However, the Dynamic Calibrator tool needs a USB 3 connection.  Even if the Android phone has a USB 3 OTG micro-size connector that you could connect the camera's USB 3 cable to, I do not know if the Dynamic Calibrator will work through an OTG type connector.

MartyG-RealSense, 2021-01-26:
Can you confirm please whether you are satisifed for the case to be closed?  Thanks very much!

closed 2021-02-02 by MartyG-RealSense