realsenseai/librealsense #8172: Announcement: RealSense Depth Module D450 + Vision Processor pack now in the RealSense Store

MartyG-RealSense, 2021-01-15:
Hi everyone,

A package of RealSense Depth Module D450 + Vision Processor board is now available in the official RealSense Store.

https://store.intelrealsense.com/buy-intel-realsense-depth-module-d450-and-d3-board-bundle-1.html

sam598, 2021-01-16:
Under Tech specs it says:
`RGB Sensor Resolution and Frame Rate | 1920 x 1080 at 30 fps`

Is that a misprint? Or does it have a different color camera than the D455?

agrunnet, 2021-01-16:
Misprint.  Thanks for informing us.

Mathias-Moon, 2021-01-18:
Thats great news. Why is this bundled with a D3 Vision processor, and not the D4 vision processor? - D455 comes with D4 as does the other modules. Is this too a misprint too? If not, is there a difference in performance compared to D455?

RealSenseCustomerSupport, 2021-01-19:
----------------------------------------------

Sorry, this is another typo. There is no D3 VPU board, actually it is called V3 version of D4 VPU Board which is for D450 module.

Thanks

closed 2021-01-27 by MartyG-RealSense