realsenseai/librealsense #8173: L515 - The drivers for this device are not installed. (Code 28)

scottrpaterson, 2021-01-16:
I've tried 4 Windows 10 computers using USB 3.0. Same error every time.

[image]

I've tried the firmware updater, but it always says "there are no connected devices".

The RealSense Viewer does not detect any device.

As a side note, I have the D435i and it works fine.

ev-mp, 2021-01-18:
@scottrpaterson , can you post the device info from [USB Device View](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview) ?
[image]

scottrpaterson, 2021-01-22:
@ev-mp 

Please find the requested information below:

[Port2]  :  USB Composite Device

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            13
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&36179869&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "Intel(R) RealSense(TM) 515"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: High (is SuperSpeed or higher capable)
Device Address:                    0x13
Open Pipes:                           2

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0210
bDeviceClass:                      0xEF
*!*ERROR:  Multi-interface Function code 239 used for device with no IAD descriptors
bDeviceSubClass:                   0x02
*!*ERROR:  bDeviceSubClass of 2 is invalid
bDeviceProtocol:                   0x01
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x8086 = Intel Corporation
idProduct:                       0x0B3D
bcdDevice:                       0x1034
iManufacturer:                     0x01
     English (United States)  "Intel(R) RealSense(TM) Camera"
iProduct:                          0x02
     English (United States)  "Intel(R) RealSense(TM) 515"
iSerialNumber:                     0x03
     English (United States)  "00000000F9441280"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0200 = 0x200 max bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0200 = 0x200 max bytes
bInterval:                         0x00

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0029  -> Validated
bNumInterfaces:                    0x02
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0xC0  -> Self Powered
MaxPower:                          0x37 = 110 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x00
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0xFF  -> Interface Class Unknown to USBView
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0200 = 0x200 max bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0200 = 0x200 max bytes
bInterval:                         0x00

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x01
bAlternateSetting:                 0x00
bNumEndpoints:                     0x00
bInterfaceClass:                   0xFF  -> Interface Class Unknown to USBView
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x5F
*!*ERROR:  No String Descriptor for index 95!

          ===>BOS Descriptor<===
bLength:                           0x05
bDescriptorType:                   0x0F
wTotalLength:                      0x0032
bNumDeviceCaps:                    0x03

          ===>SuperSpeed USB Device Capability Descriptor<===
bLength:                           0x0A
bDescriptorType:                   0x10
bDevCapabilityType:                0x03
bmAttributes:                      0x00
wSpeedsSupported:                  0x0C
  -> Supports high-speed operation
  -> Supports SuperSpeed operation
bFunctionalitySupport:             0x03 -> lowest speed = SuperSpeed
bU1DevExitLat:                     0x0A -> less than 10 micro-seconds
wU2DevExitLat:                     0x00C0 -> less than 192 micro-seconds

          ===>USB 2.0 Extension Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x10
bDevCapabilityType:                0x02
bmAttributes:                      0x00000006  -> Supports Link Power Management protocol

          ===>Platform Capability Descriptor<===
bLength:                           0x1C
bDescriptorType:                   0x10
bDevCapabilityType:                0x05
bReserved:                         0x00
Platform Capability UUID:          D8DD60DF-4589-4CC7-9CD2-659D9E648A9F
00 00 03 06 44 02 20 00

ev-mp, 2021-01-25:
@scottrpaterson , thank you for the info. The device is indeed recognized as USB2 only.
And if the device FW hasn't been upgraded recently then it still would require USB3 to operate.

You mention that you've tried different Win10 PCs as well as  switched cables with no avail, so that's not a good sign.
Therefore I would propose three additional tests -
- USB cable insertion - the enumeration type will end as USB2 if the cable is connected not fast enough. Make sure to connect the camera fast and firmly.
- The order of connections: connect the USB cable to the device first and then to PC.
- Try to connect using self-powered USB3 hub (may apply if all PCs are of the same make)

Additionally - try to connect non-realsense device to these ports on PC to verify those are indeed configured as USB3.

scottrpaterson, 2021-01-25:
@ev-mp 

- I have tried plugging the camera fast.
- I connected the USB to the camera first and PC second.
- I have not tried a USB hub, I don't have one to test, but considering how many different laptops and desktops I have tired, I think I can rule that out.
- When I plug in another device, in the exact same port such as a USB 3.0 memory stick, it correctly shows in USB Device View as 3.0.

Can you provide the Windows Driver .inf file so I can manually try installing it?

scottrpaterson, 2021-01-28:
@ev-mp - looking forward to your reply on this topic.

ev-mp, 2021-01-28:
@scottrpaterson , the cameras use standard Windows UVC driver. There are no extra drivers required.
Again, the issue is not lack of a driver but mechanical/electronic issue during plugin that prevents the device from completing USB3 enumeration correctly. 

If you've exhausted all options to troubleshoot it locally ( I still would perform the powered USB hub test), so the problem might indeed a faulty unit/connector. Thus the next step is to proceed to https://www.intelrealsense.com/support/ and check for support/replacement.
Adding @RealSenseCustomerSupport on notice as well.

RealSenseSupport, 2021-01-28:
@scottrpaterson 

I agree with @ev-mp on that this seems to be more of a faulty unit and would require a RMA process.  You can work with the intelrealsense.com webstore for return/replacement if that is the location in which you purchased the L515 camera from or if you purchased from a distributor, you can notify that distributor that you are interested in RMA.

closed 2021-02-04 by scottrpaterson

JZhu1208, 2021-06-14:
Hello, I encountered the same problem, how did you solve it? Thank you very much

scottrpaterson, 2021-06-14:
I didn't solve it, I just returned the device. I think their was a problem
with the firmware on mine... maybe yours as well.