realsenseai/librealsense #8177: Realsense Viewer SEH exception for L515 (possible duplicate of #7435)

heymikid, 2021-01-17:
Hi,

I have the same issue that was closed (#7435) with SEH exception, access violation of the RealSense Viewer on Windows 10 Pro, build 19042.

I first tried the stand-alone viewer (without even having plugged in the camera for the first time, by the way), then I plugged in the camera, and I saw that drivers were installed correctly (I see 515 Depth and 515 RGB cameras in Device Manager). Following #7435 I installed the full SDK and used the viewer that was installed (hoping for a better outcome). (v2.41.0), but I got the same error.

I have no duplicate cameras in Device Manager. However, I do see the computer's integrated camera there.

All I wanted was a quick comparison of image quality versus another LIDAR before I dive into this one. #7435 mentioned that his python scripts worked. Is there a ready-made python script that is a "viewer" that I can run quickly to assess this product?

heymikid, 2021-01-17:
I see there is a firmware update, but I am not going to attempt it until we resolve this.

Nir-Az, 2021-01-20:
Hi @heymikid ,

I tried to reproduce the issue with the same SW + FW version and could not.

I can suggest this steps:
*  Try to update the FW using [rs-fw-update](https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update) tool and try again. 
* Before connecting the L515 device please set on "Settings" a "Debug" log level and write it to a file.
Than reproduce the issue and share the log file, it might give us a hint regarding the issue you are facing.

[image]

RealSenseSupport, 2021-01-29:
Hi @heymikid 

Were you able to get things working or able to get log files?  We cannot reproduce this issue.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-03-17 by RealSenseSupport