realsenseai/librealsense #8179: Not able to run the code it raises error everytime Traceback (most recent call last):   File "C:\Users\user\Desktop\sunil\librealsense-master\wrappers\python\examples\align-depth2color.py", line 28, in <module>     profile = pipeline.start(config) RuntimeError: Couldn't resolve requests..   I am using Lidar L515 camera

sunilsai7404, 2021-01-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version |   {Win (10)  | 
                                    | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
When ever run the code it is showing this error i have gone through all the issues but i haven't got solution.

issue: 
Traceback (most recent call last):
  File "C:\Users\user\Desktop\sunil\librealsense-master\wrappers\python\examples\align-depth2color.py", line 28, in <module>
    profile = pipeline.start(config)
RuntimeError: Couldn't resolve requests

RealSenseSupport, 2021-01-28:
Hi @sunilsai7404 

With regards to the Python examples, they originally were setup for the D400 camera products and some of them have to be modified for the L515 camera.  The resolutions that are outlined in this particular python example at least for the RGB color stream by default is not supported on L515.  So one would have to modify the resolutions in the example to a supported resolution for L515.

Nir-Az, 2021-02-23:
Hi @sunilsai7404 ,
On latest release of librealsense2 SDK [2.42.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0)  the python examples were updated to support L515 devices.

try taken the example from there, it should work for you.

closed 2021-03-27 by RealSenseSupport