realsenseai/librealsense #8183: Record Multiple Cameras with python

brunovollmer, 2021-01-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 + T265 | 
| Firmware Version                   |05.12.09.00  | 
| Operating System & Version |   Ubuntu 20.02  | 
| Kernel Version (Linux Only)    |  5.4.0-56-generic                                        | 
| Platform                                 | PC  |
| SDK Version                             | legacy / 2.40.0 |
| Language                            |  Python                        | 
| Segment			|  Robot                       | 

### Issue Description

Hey,

my goal is to record synced footage from my multicam setup, which consists of a D435 and a T265 camera. I used this [issue](https://github.com/IntelRealSense/librealsense/issues/2882) as a starting point for my record script (attachted below). When I run the script I receive the following error: 
```
Traceback (most recent call last):
  File "scripts/record.py", line 133, in <module>
    main()
  File "scripts/record.py", line 127, in main
    record(args.output, args.time, args.width, args.height)
  File "scripts/record.py", line 81, in record
    depth_sensor.set_option(rs.option.inter_cam_sync_mode, 1)
RuntimeError: object doesn't support option #42
```
Anybody an idea how to fix this? 

Additionally I'm wondering if it is possible to record in the same .bag file?

**Record Script:**

```
def find_cameras(ctx):
    devices = ctx.devices

    data = {}

    for dev in devices:
        name = dev.get_info(rs.camera_info.name)
        serial_number = dev.get_info(rs.camera_info.serial_number)

        data[name] = serial_number
        print(
            f"found camera with name: {name} and serial number: {serial_number}"
        )

    return data

def record(output_path, record_time, width, height):
    context = rs.context()
    camera_data = find_cameras(context)

    base_path = '/'.join(output_path.split("/")[:-1])
    file_name = output_path.split("/")[-1]
    
    cfg_depth = rs.config()
    cfg_depth.enable_record_to_file(f"{base_path}/depth_{file_name}")
    cfg_depth.enable_stream(rs.stream.depth,
                        width=width,
                        height=height,
                        format=rs.format.z16,
                        framerate=30)
    cfg_depth.enable_stream(rs.stream.color,
                        width=width,
                        height=height,
                        format=rs.format.bgr8,
                        framerate=30)
    cfg_depth.enable_device(camera_data["Intel RealSense D435"])

    pipe_depth = rs.pipeline()

    cfg_pose = rs.config()
    cfg_pose.enable_record_to_file(f"{base_path}/pose_{file_name}")
    cfg_pose.enable_stream(rs.stream.pose)
    cfg_pose.enable_device(camera_data["Intel RealSense T265"])

    pipe_pose = rs.pipeline()
            
    depth_profile = pipe_depth.start(cfg_depth)
    pose_profile = pipe_pose.start(cfg_pose)

    depth_sensor = depth_profile.get_device().first_depth_sensor()
    depth_sensor.set_option(rs.option.inter_cam_sync_mode, 1)

    pose_sensor = pose_profile.get_device().first_pose_sensor()
    pose_sensor.set_option(rs.option.inter_cam_sync_mode, 3)

    time.sleep(record_time)

    pipe_depth.stop()
    pipe_pose.stop()

    del pipe_depth
    del pipe_pose

    del cfg_depth
    del cfg_pose
```

sam598, 2021-01-20:
inter_cam_sync_mode is used to synchronize RealSense depth cameras that have been wired together. See this article for more information:
https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-3-camera-connections

Unfortunately the T265 does not support any kind of hardware synchronization. If you try to set inter_cam_sync_mode on the T265 you will get an error.

If you need a precise synchronization between the D435 and T265 you will need to find some way to align them in software, which is not an easy task to do reliably. If you have a D435i there may be a way to align the IMU data with the T265 pose data.

brunovollmer, 2021-01-29:
My solution currently is to record both cameras in two separate depth files and then find the closest pose frame timestamp for each color + depth frame timestamp. This works fine but seems kind of hacky to me. Would still be interested in a more "developed" solution if there exists one.

RealSenseSupport, 2021-02-19:
@brunovollmer Can rosbag record function meet your requirement? It can record all currently available stream 'topics' using its -all condition. You can launch each camera independently in a separate ROS 'terminal' window, and the collective topics from all of the camera sources should get captured by the 'all' function of rosbag record. For more details, you can refer to the link in  https://wiki.ros.org/rosbag/Commandline#rosbag_record

brunovollmer, 2021-02-27:
This would be a workaround but I would prefer a single script to record.

RealSenseSupport, 2021-04-06:
@brunovollmer Sorry at present we don't have such plan to add new feature to record multiple cameras into one bag file. Please keep using the workaround you're using now. Thanks for your understanding!

RealSenseSupport, 2021-04-20:
@brunovollmer Any other questions about this? If no, will close it accordingly. Thanks!

alkasm, 2021-04-20:
As a related aside to this issue, the specific `RuntimeError: object doesn't support option #42` error seems to be due to setting `inter_cam_sync_mode` after the pipeline has started. I believe that this was previously "supported", but didn't actually set anything until the device was stopped and started again (per other issues on this repo), but now it actually throws this error if you attempt it at runtime. If you set it before you start the pipeline, e.g. if you set it on the device from a context, then it seems to be ok. For example:

```python
ctx = rs.context()
(device,) = ctx.devices
sensor = device.first_depth_sensor()
sensor.set_option(rs.option.inter_cam_sync_mode, 1.0)
pipeline = rs.pipeline(ctx)
profile = pipeline.start()
```

RealSenseSupport, 2021-04-25:
@alkasm Thanks for your sharing! 
@brunovollmer Do you get chance to try what @alkasm suggested? Looking forward to your update. Thanks!

RealSenseSupport, 2021-05-12:
@brunovollmer Any other questions about this ticket? Could you please update? Thanks!

brunovollmer, 2021-05-13:
No thanks

closed 2021-05-13 by brunovollmer