realsenseai/librealsense #8184: No module named realsense_device_manager

Oldog, 2021-01-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.12.09.00） | 
| Operating System & Version |   {Ubuntu 18.04} | 
| Kernel Version   |  (5.4.0-58-generic) |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.40 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {VR }                          | 

### Issue Description
I'm trying to run the [box-dimensioner-multicam-demo](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py). But I got an error 'No module named realsense_device_manager'. 

I have installed the pyrealsese2 so what's wrong? Or there's any other way that I can get the intrinsic parameters?

MartyG-RealSense, 2021-01-18:
Hi @Oldog   Which Python version are you using, please?   The documentation for the **box_dimensioner_multicam** example states in its **Requirements** section that it does not work with **Python 2.7**.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

Sometimes an installation of the pyrealsense2 wrapper may seem to have been successful but it is targeting Python 2.7. instead of Python 3.x.

Oldog, 2021-01-19:
> Hi @Oldog Which Python version are you using, please? The documentation for the **box_dimensioner_multicam** example states in its **Requirements** section that it does not work with **Python [...]

The python version is 3.7.  I follow the document to install the pyrealsense2 using the pip:

```
pip install pyrealsense2
```

MartyG-RealSense, 2021-01-19:
Let's check which Python version pip is targeting.  Please input the command below and let me know what it returns.

**pip -V**

MartyG-RealSense, 2021-01-26:
Hi @Oldog   Do you require further assistance with this case, please?   Thanks!

Oldog, 2021-01-30:
> Hi @Oldog Do you require further assistance with this case, please? Thanks!

Sorry for late reply. I use the conda enviroment and the pip version is 19.3.1, as shown below.

[image]

Oldog, 2021-01-30:
> Hi @Oldog Do you require further assistance with this case, please? Thanks!

Hi @MartyG-RealSense , do you have any ideas about getting intrinsic parameters of each frame when reading the bag file?There seems an API to do that but I could not find it.

MartyG-RealSense, 2021-01-30:
hi @Oldog   There was a past case where somebody tried to achieve reading the intrinsics from a bag with Python scripting but it apparently didn't work out, as they got the same intrinsic values even if they read a different bag file.

https://github.com/IntelRealSense/librealsense/issues/7309

There was another case in which the RealSense user in that case stated that they achieved it only with a laborious process involving creating a **dictionary**.

community.intel.com/t5/Items-with-no-label/How-to-save-load-intrinsics-to-from-file-python/m-p/435340#M2980

Regarding your earlier comment today about conda and pip: are you still having dififculty with **No module named realsense_device_manager** please?

Oldog, 2021-01-31:
> Hi @Oldog There was a past case where somebody tried to achieve reading the intrinsics from a bag with Python scripting.
> 
> #7309
> 
> Regarding your earlier comment today about conda and pip: are you still having dififculty with **No module named realsense_device_manager** please?
> 
> Re-reading your case from the start again, you asked if there is an alternative method to pyrealsense2 to obtain instrinsics. You can query intrinsics and extrinsics **rs-enumerate-devices -c**
> 
> https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices
> 
> Alternatively, you can instead query intrinsics and extrinsics with the **rs-sensor-control** program.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control
> 
> The discussion in the link below demonstrates querying with rs-sensor-control. Whilst the discussion uses a RealSense SR300 camera, the process is the same for the 400 Series cameras.
> 
> #5134

Thanks for the info. The device_manager issue is not solved yet. The error is shown below:
[image]

You can also find the code in the figure.

MartyG-RealSense, 2021-01-31:
There is not much available information on performing a pip install with Conda for Python 3.  There was a website that suggested using **pip3** to install the wrapper on Python 3, and then set the PYTHONPATH to the computer location where Conda is installed.

pip3 install pyrealsense2

export PYTHONPATH=$PYTHONPATH:/root/anaconda3/lib/

https://blog.csdn.net/lun55423/article/details/100177030

Oldog, 2021-02-07:
No further questions. @MartyG-RealSense  Thank you!

closed 2021-02-07 by Oldog

MartyG-RealSense, 2021-02-07:
You are very welcome @Oldog   :)   Thanks for the update!