realsenseai/librealsense #8186: Camera can get locked with old fw version

kenmccormackagility, 2021-01-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 and D435 |
| Firmware Version  | 05.12.10.00 |
| OS | Ubuntu 18.04 |
| Platform | PC |
| SDK Version | 2.41.0 |
| Language | shell scripts |
| Segment | Robot |

### Issue Description (feature request) 
 The D435 is downgraded to 5.12.03.00 for older release of our robot.  It is upgraded to 5.12.10.00 for latest release.  

After several cycles of up and down versioning the camera will report "device is locked for update" and will not accept an update.  This is expected.   The issue is that the camera is stuck at the older version of 5.12.03.00.   the FW-update tool reports "Use firmware version higher than: 05.12.10.00"    It will not accept version 5.12.10.00.

The correct thing I believe is for the camera to accept any version that is **equal to or greater** than the highest_fw_version that has been applied to the camera.   This would allow me to get the camera from 05.12.03.00 to 05.12.10.00 which is the version of our current release.   As it is the camera has to be removed and replaced with a new unit get the correct version on the robot.

MartyG-RealSense, 2021-01-19:
Hi @kenmccormackagility   There was a similar case in November 2020.  

https://github.com/IntelRealSense/librealsense/issues/7783

The eventual successful solution in that particular case was to wait for the release of a newer firmware version than the one that the camera believed was currently installed and then install it.

https://github.com/IntelRealSense/librealsense/issues/7783#issuecomment-737898217

MartyG-RealSense, 2021-01-19:
Hi @kenmccormackagility   There was a similar case in November 2020.

https://github.com/IntelRealSense/librealsense/issues/7783

The eventual successful solution was to wait for the release of a newer firmware version than the one that the camera believed was currently installed and then install it.

https://github.com/IntelRealSense/librealsense/issues/7783#issuecomment-737898217

kenmccormackagility, 2021-01-22:
> #7783 (comment)

Thanks for the link to the similar issue (#7783)   Our issue is  similar
but perhaps is a feature request rather than a bug.  The realsense is
deployed in a commercial robot and waiting for an upgrade is not an
option.    It seems that a camera that is locked at an older version should
always be upgradable to the newest version that has been previously
installed on the camera.  It should not be limited to newest+1.   As it is
now the camera cannot be upgraded in the field until a new release that is
newer than 05.12.10.   Being locked at an older version the camera will not
work in the robot.

A second question that is related.   Does anyone know the reason a camera
would get locked in the first place?   I've tested two other cameras
bouncing between two versions (05.12.03/05.12.08) and they seem to
indefinitely update.   Yet I have two other cameras that stopped updates
and are now locked out.

MartyG-RealSense, 2021-01-23:
Could you confirm please whether updating the camera in the RealSense Viewer instead of rs-fw-update, as suggested above, enabled the camera to be successfully updated to 5.12.10.0 please?

Is using the Viewer as a means of updating the firmware in future instead of the rs-fw-update tool an option for your commercial operation?

Regarding your second question, the official information about the 20-downgrade limit in the data sheet document for the 400 Series cameras is quoted below.

**********************

The firmware update engine does not allow infinite update cycles between older and current versions of firmware.  The engine will establish a baseline version of firmware based on the latest firmware version installed.  The engine will allow a return to a previous version or baseline version of firmware up to 20 times.  After the 20th update, the engine will only allow an update to a firmware revision higher than the baseline version.

********************

As far as I know, only a downgrade to an older firmware version than the currently installed version - or to the version considered as the camera's 'baseline' firmware - counts as one of the 20 permitted attempts, not an upgrade (e.g from 5.12.3 to the newer 5.12.8).  The origin of the 20 downgrade limit dates back to the SR300 RealSense camera model launched in early 2016.  I do not know of a circumstance that would cause the counter to reset and allow unlimited downgrades though, as the data sheet document statement represents the extent of the publicly available information about the 20-limit.

kenmccormackagility, 2021-01-26:
I have verified that the camera *does not *upgrade from 05.12.03.0 using
the realsense viewer.      It reports the following

   - Requesting to switch to recovery mode.
   -  Recovery device connected, starting update.
   - ERROR:UNKNOWN in rs2_update_firmware_cpp()  Device: 818413020943 is
   locked for update.
   -  Use firmware version higher than 5.12.10.10"

Upgrading with the realsense-viewer is an option.  However, since both the
rs-fw-update and the viewer use the same updating libraries it seems like a
command line tool can be devised that would work as good as the viewer.

Thanks for the information on the 20 cycle limitation.

MartyG-RealSense, 2021-01-26:
Thanks very much for the information @kenmccormackagility 

Though updating through rs-fw-update or the Viewer are the recommended update methods, it is also possible to check firmware information and update the firmware with scripting instructions:

https://github.com/IntelRealSense/librealsense/issues/4416

kenmccormackagility, 2021-01-27:
I have manually verified that 20 or so cycles of changing versions on a
camera will cause it to lockup and fail to accept new updates that are not
one version greater than the newest version seen on the device.   I have
also verified that if the version+1 is applied to the device that it resets
the up/down upgrade count back to zero and thus allows updates as desired
freely.

This provided a work around for the issue seen.   We use the production
versions of FW release and if a customer upgrades/down grades enough to
trigger the issue we will service the device by writing one of the "minor"
releases to the camera.  After this any version that is compatible
including the current baseline can be put back on.

The feature requested here is that the device should always be able to
return to the baseline version irregardless of the number of up/down
cycles.

This issue can be closed.

MartyG-RealSense, 2021-01-28:
Thanks so much @kenmccormackagility for the detailed update for the benefit of RealSense cmmunity members reading this case in future!  I am very pleased that you reached a satisfactory solution.

closed 2021-01-28 by MartyG-RealSense

liuxiaodong528, 2022-03-01:
> 

Hello, what is the purpose of the 20 times  limit to downgrading？

MartyG-RealSense, 2022-03-01:
Hi @liuxiaodong528  There is not an official explanation, but unofficially my understanding is that it is a protective measure to stop multiple changes to the camera firmware being made by an unauthorized source such as harmful 'malware' software.  Once the downgrade limit of 20 is reached then from that point onward, the camera can only be upgraded to a newer firmware version than the currently installed version.