realsenseai/librealsense #8187: D435i Not connecting to RPI 4

HFeustel, 2021-01-19:
Hi there,

I am currently working with a D435i camera on an RPI4. I am running the newest version of Raspberry Pi OS on the Pi and have successfully installed the SDK. I hooked the camera up to my 3.0 USB port, however when I run the python-tutorial-1-depth.py test I get a "No device connected" error. If I run lsusb command, the device is found and when I plug the camera in and run a dmesg command, it gives the following output: 
[image]

*Note the code compiles, but it does not connect. Also, I have used this camera and USB cord to connect to a Linux VM and know that it is possible to connect. Am I missing something in how the RPI accesses the pipeline?

MartyG-RealSense, 2021-01-19:
Hi @HFeustel   If you are using a Linux VM with librealsense then you may encounter problems due to difficulty in interfacing with USB 3.   If you do need to install librealsense on a VM, Intel's documentation suggests using **VMWare Workstation Player**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

HFeustel, 2021-01-19:
Hi @MartyG-RealSense thanks for the reply!

I have already done that and I am fine when connecting to the camera. For more context, I am using the RealSense camera for a robot to do image detection. For testing, I used my Windows machine with a linux VM in VMWare and developed my code and everything worked. Now, I am going to copy my code to an RPI because that is the computer on the robot. However, when I compiled the realsense library and ran a test file, it is saying I am not connected to a device even though the camera is plugged in. The image I put in the original post was the response to the dmesg command on the RPI to see if the machine was connecting to camera (which it is), but my code is saying that it isn't.

HFeustel, 2021-01-19:
@MartyG-RealSense,

I deleted the library and recompiled it on the RPI and it works now. The only tweak I made was that when I copied the udev rules, I made sure the camera was unplugged this time. Last time, I set the udev rules with the camera plugged in.

closed 2021-01-19 by HFeustel

MartyG-RealSense, 2021-01-19:
Thanks very much for the update @HFeustel  - great news that you succeeded!  Thanks for sharing the solution that you used.