realsenseai/librealsense #8188: Does decimation filter alter the width and height of the images ?

zc08, 2021-01-19:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Ubuntu 18.04.3 |
| Kernel Version (Linux Only) | 5.4.0-48-generic |
| Platform | PC |
| SDK Version | 2.36.0 |
| Language | C/python |
| Segment | Robot |

### Issue Description
I'm trying to down sample the frames using the decimation filter, using the following code:

```   
        rs2::decimation_filter dec_filter;
        frames = frames.apply_filter(dec_filter); 
```
However, it seems that the `color` and `depth` frames I get after decimation, have the same width and height as the original frames.

Is my usage correct or what could be the problem ? Thanks in advance !

zc08, 2021-01-19:
It seems the decimation filter alters the width and height of the depth frame, but not the color frame.

closed 2021-01-19 by zc08

MartyG-RealSense, 2021-01-19:
Hi @zc08   I had written a reply just before you closed the case, so I will post it for the future reference of other RealSense users.   :)

*********************

The decimation filter scales height and width proportionally to maintain aspect ratio, as described in the post-processing documentation section linked to below,

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter

This documentation page also provides an example of applying a decimation filter in C and C++ languages:

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#using-filters-in-application-code

For the Python language, a simple method of decimation is suggested here:

https://github.com/IntelRealSense/librealsense/issues/2356#issuecomment-418750368