realsenseai/librealsense #8189: Intel RealSense Viewer freezes with D415 after a few seconds of streaming

krkkvpl, 2021-01-19:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|

Hi,

I'm having problems using Intel RealSense D415 with the RealSense viewer. The viewer detects the camera fine and allows me to open up streams to see depth/RGB info, but freezes after 2-20 seconds of streaming. After freezing, I'm sometimes able to close down the stream. However, upon reopening a stream I get:

> Invalid Value in rs2_open_multiple(sensor:0000020AD6FF0750, profiles:0000020AE8825B88, count:1):
> 
> Failed to resolve the request: 
> 	Format: Z16, width: 1280, height: 720
> 
> Into:
> Device is already streaming!

To reopen a stream I have to close down the program, sometimes forcing it to quit (sometimes it closes down nicely without forcing). Every now and then, the RealSense viewer freezes at the flash screen. To resolve this, I have to disconnect the camera, open up the viewer, close the viewer, reconnect the camera and then open up the viewer again.

I have tried turning the advanced mode on and off. This does not change the ill behaviour at all.

I have also encountered this error twice upon opening a stream. Might be related:

> Unhandled exception escaping from a worker thread!
> Error type: Access Violation!

I would be glad to provide any information you need to resolve this issue.

Best regards,
Joel Lindfors

MartyG-RealSense, 2021-01-19:
Hi @krkkvpl   As a starting point in investigating your case, could you check whether disabling an option called **GLSL** in the RealSense Viewer's settings makes a difference to your problem, please?

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

krkkvpl, 2021-01-19:
Hi Marty!

I tried disabling GLSL for both rendering and processing. The freeze still occurs.

I will add that in general the freezing occurs much faster with the depth sensor on.

Thank you for the help so far.

Br,
Joel

MartyG-RealSense, 2021-01-19:
Thank you very much for checking GLSL.

Next, may I ask which method you used to install the SDK on Windows.   Did you use the automated installer file from the SDK **Releases** page or build Windows from source code with **CMake**?

krkkvpl, 2021-01-19:
I used the automated installer from here:
https://www.intelrealsense.com/sdk-2/

MartyG-RealSense, 2021-01-19:
For future SDK updates in Windows, I would recmmend obtaining the installer file from the **Releases** page at the link below.  The file is in the **Assets** file list at the bottom of the information listing for each version.

https://github.com/IntelRealSense/librealsense/releases

[image]

Which firmware driver version do you currently have installed in your camera please?  For SDK **2.40.0** the Releases page recommends firmware version **5.12.9.0**.

If you are not already using 5.12.9.0 or newer in your camera then you can update the firmware by going to the **More** option near the top of the options side-panel of the 2.40.0 Viewer and selecting **Install Recommended Firmware** from its menu.

[image]

krkkvpl, 2021-01-20:
Hi,

I had already updated the firmware to the recommended one before I made this post. I have now updated the Viewer to 2.41.0.2657. The problem persists.

I also got this error now:

> Backend in rs2_get_option_range(options:0000026A5DA21D30, option:Enable Auto White Balance, min:0, max:0, step:4.59037e-41, def:2.32692e-35):
> ks_control->KsProperty( reinterpret_cast<PKSPROPERTY>(&node), sizeof(node), &description, sizeof(KSPROPERTY_DESCRIPTION), &bytes_received) returned: HResult 0x800706bf: "The remote procedure call failed and did not execute."

After which opening a stream returns a "No Frames Received!" or simply a split second of stream from the camera and then freeze.

I will be glad to provide further information to resolve this issue.

krkkvpl, 2021-01-20:
I went ahead and tried using the camera in a with pyrealsense2. Here I encountered a similar error when using a openCV example code from here:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

After running the program, I see the depth data and the RGB data, but after a similar amount of time as with the Viewer, I get this error

[image]

I did also get this weird behaviour in some runs of the program, where I only saw shades of blue in the depth image, as if it wasn't gathering information from all of the sensors.

I hope this helps!

MartyG-RealSense, 2021-01-20:
The next question that I would ask is whether your Windows computer has an AMD processor CPU.  The 400 Series cameras should be able to work with any Intel or Arm processor, but the results from using the camera on a computer with an AMD processor may be more unpredictable.  I have seen a small number of reports of the cameras working with AMD Ryzen and Threadripper processors.

krkkvpl, 2021-01-21:
My computer runs on Intel Core i7-6800K.

closed 2021-02-05 by MartyG-RealSense