realsenseai/librealsense #8191: ImportError: No module named pyrealsense2

lamngoctam, 2021-01-20:
I am using Jetson Xavier AGV, Ubuntu 18.04, and t265 and D435i camera.

I checked the issue and got the potential solution from @lieuzhenghong
"https://github.com/IntelRealSense/librealsense/issues/7722".
However, the issue could not be solved.

I tried to run the example from "https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_example.py"
the error is below:
```
Traceback (most recent call last):
  File "t265_ex1.py", line 11, in <module>
    import pyrealsense2 as rs
ImportError: No module named pyrealsense2
```
Does anyone have the same problem? and any advice? Thanks!!!

yosoufe, 2021-01-27:
I have compiled for Jetson Nano with this instruction and it worked. You may try it: https://github.com/yosoufe/SelfStudyRobotics/blob/master/hardware/realsense.md#compile-for-jetson-nano

You may also have look at the discussions here: https://github.com/IntelRealSense/librealsense/issues/7660

I hope it  helps

RealSenseSupport, 2021-02-03:
@yosoufe Thanks very much for your sharing!
@lamngoctam Did you get chance to try what suggested by yosoufe? Looking forward to your update. Thanks!

lamngoctam, 2021-02-04:
@yosoufe thanks so much. I solved the problem!

closed 2021-02-04 by lamngoctam