realsenseai/librealsense #8192: D455 depth accuracy

ChenXinThales, 2021-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455/ D415  | 
| Firmware Version                   | 5.12.7 | 
| Operating System & Version |  Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.38.1  | 

### Issue Description
I have used Depth Quality Tool to test my D455 and D415 as below, it seemed that D415 RMS Error was larger than D455.
[image]
[image]
Howerver, when I tested a person at about 1m, with different presets, D415 always showed more details. Here are the results with Default preset.
[image]
[image]
[image]
So, can I draw a conclusion that D415 depth accuracy is better than D455 at 1m? 
And is there a curve for D455 like this?
[image]

MartyG-RealSense, 2021-01-20:
Hi @ChenXinThales   There is not an equivalent RMS error chart for D455.  However, the D455 model has 2x the depth accuracy over distance of the D435.  So at 6 meters distance the D455 has the same depth accuracy as the D435 has at 3 meters.

The D415 meanwhile has around 2x improved RMS error over distance than the D435 model, as indicated by the lower green curve representing the D415 on the chart above.  The D455's enhanced accuracy over distance may account for the difference between the RMS value for it and D415.

In regard to D415 showing more details: the D415 has a minimum depth sensing distance of 0.3 meters, whilst the D455 has a minimum distance of 0.4 meters.  So it may be that the extra detail on the D415 image is near-range detail that is below the D455's minimum depth observing distance.  

You can test whether this is the case in the RealSense Viewer without moving the camera by using an option called **Disparity Shift**.  Increasing the Disparity Shift value reduces the minimum distance (MinZ) of the camera (enabling it to get closer to objects), whilst reducing the maximum depth sensing distance (MaxZ).  This option can be found in the following location in the options side-panel:

**Stereo Module > Advanced Controls > Depth Table > Disparity Shift**

[image]

ChenXinThales, 2021-01-20:
No, the picuture I posted was a man's side face,  I mean we can tell the outline of the face in D415, like nose, mouth. But we can only see a plane in D455
[image]

MartyG-RealSense, 2021-01-20:
I ran tests in the 3D point cloud mode with changing the prset, as you mentioned that you were using Default.  I found that you may obtain worse depth results if you change the preset whilst the stream is active, whereas the depth image was much better if setting the preset whilst the stream was disabled and **then** enabling the stream.

Have you tried setting the Default preset whilst the stream is Off before activating the stream, please?

ChenXinThales, 2021-01-20:
Yes, I have tried, no improvement

MartyG-RealSense, 2021-01-20:
Does the detail improve if you move the D455 camera a little closer to the human?

Is it practical for you to increase detail on the image by mapping the RGB to the depth points by going to the **Texture** option and selecting the **Color** option?

[image]

ChenXinThales, 2021-01-21:
When I moved forward, the detail improved little. Here are my test results
Move forward
[image]
Color D455
[image]
Color D415
[image]
Is D415 depth accuracy better than D455 at 1 meter? Or do I still need to adjust D455?

MartyG-RealSense, 2021-01-21:
I performed further point cloud comparison tests of D455 vs D415 and had very similar results to the differences between your own D415 and D455 images, primarily the larger field of view on D455 (hence a greater amount of detail on the D455 image).  As with your image set, my test images also had warmer colors on D415.

As far as I can tell, there is therefore not a problem with your cameras and the results received are correct for the particular camera models being used.

The color shades can be made warmer by manually adjusting the **White Balance ** slider under the Viewer's RGB controls.

[image]

Changing the **Depth Units** of the D455 from the default of **0.001** to a smaller **0.0001** yielded a small improvement in the amount of depth detail that was produced.

[image]

agrunnet, 2021-01-21:
This is consistent with what we expect. The d435 and d455 do have fewer pixels spaced across a wider FOV.  So when we compare x-y minimum measurable object size it will be worse than D415.  The z axis resolution will be similar for d455.

ChenXinThales, 2021-01-22:
Thank you very much.

closed 2021-01-22 by ChenXinThales