realsenseai/librealsense #8193: D415 FW update error HResult 0x8007001f: "A device attached to the system is not functioning."

lch-dol, 2021-01-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D410 | 
| Firmware Version                   | 5.12.7.100 and 5.12.9.0 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.41.0 | 
| Segment			| Firmware update | 

### Issue Description
When I update the firmware version of my realsense cameras (D415 and D410) the update tool (SDK 2.41.0 Intel.RealSense.FW.Update.exe) starts off with an error but then finishes:
[image]
The error 0x8007001f seems to have be reported a few places but I can't seem to find a specific reason. My question is: although the FW update seems to be OK does this error hint at something actually being wrong?
I have tried different computers, different FW version (as listed above), and different cameras including a brand new one, but i always get the error.

Best regards,
Lars

MartyG-RealSense, 2021-01-20:
Hi @lch-dol  I studied your case carefully.  Given that the firmware update apparently succeeds and you have had the same result with different computers, firmwares and cameras (presumably each with its own USB cable), this would suggest by a process of elimination that the issue is related to the **rs-fw-update** program.  This is because the program is the single common factor between all of your attempts with different computers and cameras.

I am inclined to believe that the message **A device attached to the system is not functioning** represents a momentary 'hiccup' that the firmware updater recovers from to complete the update.  I would recommend testing the updated cameras thoroughly in the **RealSense Viewer** program.  If you do not encounter significant problems during these tests then it is likely that the firmware update was successful.

You can verify in the Viewer which firmware version is currently installed by left-clicking on the 'Info' option at the top of the Viewer's side-panel.

You can also update firmware in the Viewer with a firmware file instead of rs-fw-update by going to the **More** side-panel option, selecting **Update Firmware** from its menu and then browsing the pop-up file selector to the location of the firmware file.

[image]

lch-dol, 2021-01-22:
Hi @MartyG-RealSense.
Thanks for the quick answer!
We have tested the camera and everything seems to be OK. The only thing we have seen during our test, is that sometime (most of the times) after a FW update the camera must be disconnected from the computer and then reconnected before we can do a second FW update.
I just wanted to make sure that it wasn't something you knew could cause a "hidden" problem. For now we will continue by simply ignoring the first error.

MartyG-RealSense, 2021-01-22:
I'm very pleased to hear that you have had stable test results.  I would just add that it is preferable to perform firmware updates with the official 1 meter cable supplied with the camera if possible if you are not already doing so, and avoid performing the update over a very long USB extension cable, such as one with an **active repeater** signal booster built into it.

MartyG-RealSense, 2021-01-29:
Hi @lch-dol   Do you require further assistance with this case, please?   Thanks!

lch-dol, 2021-01-29:
Hi @MartyG-RealSense
I do not need further assistance. As I mentioned we will continue by ignoring the first error and I will return if we start experiencing any problems.
Do you want me to close the topic or should it stay open until a potential error has been found in the rs-fw-update program?

MartyG-RealSense, 2021-01-29:
If you do not need further help, it is fine to close the topic and return to it if you experience it again in future.   Thanks for the quick response!

closed 2021-01-29 by lch-dol