realsenseai/librealsense #8194: The definition of rs2_frame

zc08, 2021-01-20:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Ubuntu 18.04.3 |
| Kernel Version (Linux Only) | 5.4.0-48-generic |
| Platform | PC |
| SDK Version | 2.36.0 |
| Language | C/python |
| Segment | Robot |

### Issue Description
I'm trying to get a bit more understanding of the code of librealsense, digging into the definition of `rs2::frame` in `rs_frame.hpp`. It contains a pointer to `rs2_frame`.

However, the only thing defining `rs2_frame` I can find is in `rs_types.h`:

```typedef struct rs2_frame rs2_frame;```

which doesn't actually define the data structure.

How is `rs2_frame` defined ? 

PS: The background of this question is that, I'm trying to see if it's possible to convert `cv::Mat` to `rs2::frame`, so I could use some functions from librealsense to process `cv::Mat` data.
Thanks in advance !

MartyG-RealSense, 2021-01-20:
Hi @zc08   A diagram of the structure of **rs2_frame** can be found here:

http://docs.ros.org/en/kinetic/api/librealsense2/html/classrs2_1_1frame.html

There is a detailed discussion about converting cv::mat to rs2::frame in the link below.

https://github.com/IntelRealSense/librealsense/issues/2634

MartyG-RealSense, 2021-02-02:
Case closed due to no further comments received.

closed 2021-02-02 by MartyG-RealSense

ShaneKosieradzki, 2022-06-21:
You have linked documentation to the `rs2::frame` OBJECT but OP, and now I, are asking about the `struct rs2_frame` found [here](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_types.h#L262)

MartyG-RealSense, 2022-06-21:
Thanks very much @ShaneKosieradzki for your clarification and link.  What information are you seeking about **struct rs2_frame** please?

ShaneKosieradzki, 2022-06-21:
Hello @MartyG-RealSense thank you for your quick reply,

I would like to know the definition of the `struct rs2_frame` which seems to be the fundamental `struct` to work with the C-API. This struct does not have any definition within the codebase and I would rather not decompile the `.lib` to find the definition.

For context we are security researcher studying encryption of low-level devices. We would like to take the point data contained in a `struct rs2_frame` and encrypt it. However in order to do so we need to be able to write our encrypted data to the appropriate object, which seems `struct rs2_frame`.

Any advice or help you have in this matter would be greatly appreciated.

ShaneKosieradzki, 2022-06-21:
We have been using `get_data` as well as `get_vertices` to extract the data. We have two issues with this

1) The distinction between `get_data` and `get_vertices` is not as clear as it could be. `get_data` returns a `void*` so it seems like the raw data, and `get_vertices` returns a `vertex*`. I am not sure if these are pointing to the same memory location and just using different types to interpret the data, but that is my intuition. Is that correct?

2) After we extract and encrypt the data we have no way of putting the encrypted data back into a `frame` object. We believe that `rs2_frame` is the object we need to load our encrypted data back into a `frame`.

MartyG-RealSense, 2022-06-22:
Further research found that the SDK file **unit_tests_common.h** may be a useful reference for struct information, especially the **internal_frame_additional_data** struct linked to below.

https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-common.h#L510

In regard to your questions:

1.  A distrinction that can be made between get_data and get_vertices is that get_vertices is typically used to retrieve stream values, such as point cloud coordinates, that have been stored as vertices so that they can be accessed as individual XYZ coordinates.  A Python code snippet below shows an example of get_data and get_vertices being used in the same code block.

[image]

An example use of get_data would be to obtain the raw 16-bit depth value (uint16_t) and multiply it by the camera's depth unit scale (usually 0.001 by default on 400 Series cameras) to obtain the real-world distance in meters, as demonstrated in https://github.com/IntelRealSense/librealsense/issues/8150#issuecomment-759344866 - for example, a uintvalue of 65535 x 0.001 would give 65.535 meters in real-world distance.

3.  The RealSense SDK offers the ability to create your own custom frameset for data instead of using the SDK's frameset.    Information about doing this in C++ can be found at https://github.com/IntelRealSense/librealsense/issues/5847