realsenseai/librealsense #8196: L515 - setting stream resolution to 1024x768 creates unknow exception

eddiz, 2021-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515 } | 
| Firmware Version                   | 1.5.3| 
| Operating System & Version |   Win 10  |                                   | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.41.0 }                          | 
| Language                            | C++                        |  

### Issue Description
Setting the image resolution for depth and color streams to 1024x768 in config::enable:stream, an exception `(RS2_EXCEPTION_TYPE_COUNT(8))` is raised in `pipeline::start(config)`. The resolution should be the native resolution of the depth sensor and this option is actually available in the real-sense viewer. Setting a resolution of 640x480 works fine. This seems to be the default resolution in many examples.

Minimal code snippet to reproduce is:

`rs2::pipeline pipe;`
`rs2::config cfg;`
`cfg.enable_stream(RS2_STREAM_DEPTH, 1024, 768, RS2_FORMAT_Z16, 30);`
`cfg.enable_stream(RS2_STREAM_COLOR, 1024, 768, RS2_FORMAT_RGB8, 30);`
`pipe.start(cfg);`

A small example would be a modified version of rs-align.cpp

I haven't found a similar issue.

eddiz, 2021-01-20:
Ok, found the answer on my own. 
The color and depth stream resolutions must be set separately - e.g. color should be 1280x720.

Nevertheless, the raised exception should be `RS2_EXCEPTION_TYPE_INVALID_VALUE ` and not `RS2_EXCEPTION_TYPE_COUNT`

RealSenseSupport, 2021-01-29:
@eddiz - Thanks for the information and glad to see you were able to get things working.  For L515 the resolutions between depth and color are not the same set of supported resolutions.

Is there anything else you need on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-02-05 by RealSenseSupport