realsenseai/librealsense #8197: Depth/Color syncer with external HW-Trigger: "frame ... was not found"

Artur078, 2021-01-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (5.12.3.0) | 
| Operating System & Version |   {Linux Raspian Debian Buster  | 
| Kernel Version (Linux Only)    |  (5.4.72)                                         | 
| Platform                                 | Raspberry Pi |
| SDK Version                            |  { 2.39.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Tool }                          | 

### Issue Description
hello,
I am currently working on a project with a multi-camera setup with 2 D435i Cameras. 
Both Cameras are to be HW-synchronized by one external trigger.
The burst rate in the gelock mode has to be greater than 1.

For Each camera the depth and color frames have to be synchronized. 
I Realized this with the syncer class method, one syncer for each camera.
Instead of using top-level API pipeline and config, I use context to get direct access to the sensors
(because otherwise throwing "failed to set powerstate").
Global_Timestamp_Domain.
The sensors are opened with an assigned video_stream_profile of same framerate.
They are started with sensor.start(sync) before external Trigger-Signal may arrive.
 
After triggering, the Framesets are read out by the poll_for_frames or wait_for_frames method .
There I receive the error-statement: null pointer passed for argument ""frame_ref", (if i use the get_depth_frame method)
resp. "frame of requested streamtype was not found" (if i use frameset.first(RS2_STREAM_ANY)).
Obviously no frame was recorded.

my guess is that depth_sensor did not match appropriate color_sensor frame and returns empty frame instance.
Thank you for any provided hints that may help.

sam598, 2021-01-20:
This may not be the specific issue you are experiencing, but if you are using an external synchronization trigger by setting the "intercam sync mode" to 4 or greater (aka the "genlock" mode) only the global shutter infrared/depth cameras will be triggered. Rolling shutter and color cameras will not be synced, and instead will run at their full requested framerate.

See 2.2 here:
https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-2-principles-of-operation

This is a hardware limitation and will never be changed. If instead you use one of the D435i cameras as the master sync signal, the color cameras should be running in sync.

It is possible that the D455 will support genlock sync for global shutter color cameras in a future update (https://github.com/IntelRealSense/librealsense/issues/7502) but no timeline has been given.

MartyG-RealSense, 2021-01-20:
Thanks @sam598 for your assistance in this case!

@Artur078   Does the advice of @sam598 answer your questions or do you require further assistance?  Thanks!

Artur078, 2021-01-20:
Thanks for the answer. But when using the external trigger shouldn't be both cameras be in full slave (genlock) mode?

sam598, 2021-01-20:
@Artur078 the RealSense "cameras" contain multiple cameras that are separate hardware. Sometimes these sub-camera are global shutters, and others are rolling shutters. There are two ways to synchronize multiple cameras, either amongst themselves or through an external signal.

**Genlock through external sync signal (modes 4-255):**

D415 - No Genlock support. All cameras are rolling shutter.
D435 - Genlock support for infrared/depth cameras. No Genlock support for rolling shutter color camera.
D455 - Genlock support for infrared/depth cameras. Genlock support for color cameras are planned but no ETA.

**One RealSense camera generates a master signal, others are slaved to it (mode 1 and 2):**

D415 - Full support.
D435 - Rolling shutter color camera may not be in sync with Global Shutter infrared cameras (this may have changed).
D455 -  Full support.

Artur078, 2021-01-20:
Thanks @sam598 for the explanation. Based on it I will try to find a solution.

Artur078, 2021-01-25:
I still can not cope to synchronize depth and color frame with the syncer. 
The sync queue only contains one color_frame. 
Thank you for any help.

MartyG-RealSense, 2021-01-27:
Hi @Artur078   Do you have an update of your progress that you can provide, please?  Thanks!

Artur078, 2021-01-27:
@MartyG-RealSense I was still not able to synchronize depth and color frame with the syncer,
Only first color-frame is being retrieved.
Trying to get the depth-frame results in the exception: null pointer passed for argument ""frame_ref".

The depth-sensor is externally HW-triggered in genlock-Mode or slave-mode. 
The color-sensor is free-running or simultaniously started via GPIO.

I tried to get a single depth-frame with the external trigger signal without synchronizing to color-frame.
The depth-sensor produces frames independently of trigger-start, 
but the frames can not be retrieved either, throwing same exception as above.

I looked up in rs_frames.hpp but could not figure it out.

MartyG-RealSense, 2021-01-28:
I will require some additional time to research your question above before answering.  I do apologize and appreicate your patience.

MartyG-RealSense, 2021-01-29:
@sam598   As you have advanced knowledge of genlock, would it be possible to advise @Artur078 again on the most recent question please?  I appreciate your helpfulness very much.

https://github.com/IntelRealSense/librealsense/issues/8197#issuecomment-768496750

Artur078, 2021-01-30:
@MartyG-RealSense Thank you for your initiative.
Meanwhile I alternatively used a self-defined callback with the sensor.start()-method and STL Templates to save the frames.
Since only one frame per Trigger-Signal is demanded for my special purpose, I actually do not need timestamp-synchronizing,
polling and framequeues. The processing is fast enough for my needs.
Nevertheless I would appreciate it for future projects, if the provided SDK-Functions worked.
Thank you again.

MartyG-RealSense, 2021-01-31:
Hi @Artur078   It is excellent news to hear that you developed a solution that is satisfactoy for your needs.   Thanks again for your patience during this case.

closed 2021-02-07 by MartyG-RealSense