realsenseai/librealsense #8198: realsense-viewer: undefined symbol: glfwMaximizeWindow

grupot3, 2021-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 5.12.7.100 | 
| Operating System & Version |   Ubuntu  | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                         | 
| Platform                                 | JetsonTX2  |
| SDK Version                            | v2.31.0                        | 

### Issue Description
All the installation went smoothly, and I'm able to run the cameras using the ROS wrapper. 

However,  if I try to launch the `realsense-viewer`, I get the error 

`realsense-viewer: symbol lookup error: realsense-viewer: undefined symbol: glfwMaximizeWindow`

What should I do?

Thanks!

dorodnic, 2021-01-21:
It seems you have older version of GLFW installed and it is conflicting with the one inside librealsense. 
I would try `sudo apt remove libglfw3 libglfw3-dev`

RealSenseSupport, 2021-01-28:
Hi @grupot3, Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2021-02-17 by RealSenseCustomerSupport