realsenseai/librealsense #8199: RealSense error with Infrared frames

sandeshk1, 2021-01-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.
* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  5.4.0-58-generic                                        | 
| Platform                                 | core i7  |
| SDK Version                            |  Real Sense SDK 2.0                      | 
| Language                            |  C++                          | 
| Segment			|  Vision                          | 

### Issue Description
Error while reading infrared frames from camera and converting it to OpenCV Mat

Code snippet :

     rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        rs2::depth_frame depth = data.get_depth_frame();
        rs2::frame color = data.get_color_frame();
        rs2::video_frame ir = data.get_infrared_frame();

        const int irw = ir.get_width();
        const int irh = ir.get_height();

        // Create OpenCV matrix for ir data
        Mat irimage(Size(irw, irh), CV_8UC1, (void*)ir.get_data());

Error o/p : 

    RealSense error calling rs2_get_frame_width(frame_ref:nullptr):
    null pointer passed for argument "frame_ref"

I did not face this issue with color and depth frames while converting them to OpenCV Mat

MartyG-RealSense, 2021-01-20:
Hi @sandeshk1   The link below has a script shared by a RealSense user that successfully exports left and right RealSense infrared to cv::mat.  This should hopefully be a useful reference for your own project.

https://github.com/IntelRealSense/librealsense/issues/1480

sandeshk1, 2021-01-21:
Why doesn't the get_infrared_frame() member function throw the error ? Is it mandatory to use the rs2::config class to enable stream ?

ev-mp, 2021-01-21:
@sandeshk1  hello,
>Is it mandatory to use the rs2::config class to enable stream ?

`rs::config` is used to enable stream with `pipeline` objects
Enabling streams with sensor API is done with `sensor.open(profile)`, see [tutorial](https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L453)

>Is the error thrown as we try to query the width and height from the infrared frame ?

The error is thrown since the frameset does not guarantee frames to be available, so they shall be validated before being dereferenced:
```c++
rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
rs2::depth_frame depth = data.get_depth_frame();
rs2::frame color = data.get_color_frame();
rs2::video_frame ir = data.get_infrared_frame();

const int irw = ir.get_width();  //  is ir==nullptr ?
```

A snippet from [rs-post-processing demo](https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp#L108-L111) for reference
```cpp
rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
rs2::frame depth_frame = data.get_depth_frame(); //Take the depth frame from the frameset
if (!depth_frame) // Should not happen but if the pipeline is configured differently
    return;       //  it might not provide depth and we don't want to crash
```

MartyG-RealSense, 2021-01-28:
Hi @sandeshk1  Do you require further assistance with this case, please?  Thanks!

sandeshk1, 2021-01-29:
Thanks all .. I think the information provided is sufficient

closed 2021-01-29 by sandeshk1