realsenseai/librealsense #8201: D455 does not show IMU anymore

cansik, 2021-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    | 10.15.7 (19H114)                         | 
| Platform                                 | PC |
| SDK Version                            |  2.40.0                  | 

I have updated the D455 to FW version 05.12.10.00 and am using the RealSenseViewer from v.2.40.0 (MacOS) and it seems that the IMU Module (Motion) is not shown anymore. I can not start gyroscope / accelerometer streams from the code as well.

[image]

Is that an intended behaviour?

MartyG-RealSense, 2021-01-20:
Hi @cansik   Dorodnic the RealSense SDK Manager made reference in November 2019 to how IMU data could not be accessed yet on MacOS.

https://github.com/IntelRealSense/librealsense/issues/5261#issuecomment-554916553

The D435i, which like the D455 is equipped with an IMU, is not currently supported on Mac, as indicated by the extract below from the SDK's **Releases** page.

[image]

I checked the features **Support Matrix** document, which Dorodnic also referenced in his explanation.  IMU hardware timestamps are still listed as **(No) WIP** - Work in Progress - under the MacOS column of the document.

https://dev.intelrealsense.com/docs/supported-features-matrix

[image]

cansik, 2021-01-20:
Ok, thank you very much for clarification!

closed 2021-01-20 by cansik

MartyG-RealSense, 2021-02-15:
Hi again.  The new librealsense version **2.42.0** has introduced IMU support for MacOS.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0