realsenseai/librealsense #8202: "Incomplete video frame detected" Ubuntu 18.04

jigglyjoubres, 2021-01-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer v2.41.0) | 
| Operating System & Version |   Linux/Ubuntu(18.04)  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.0                          | 

### Issue Description
<When I open the RealSense viewer, a warning keeps opening that an incomplete video frame is detected. I checked the usb connection and it is all good and I have 3.2 next to my camera's name in the viewer.>

jigglyjoubres, 2021-01-20:
I am also using Ubuntu via PC and the USB cord that came with the camera

MartyG-RealSense, 2021-01-20:
Hi @jigglyjoubres   **Dorodnic** the RealSense SDK Manager has commented in the past about the possible causes of incomplete frames occurring:

https://github.com/IntelRealSense/librealsense/issues/927#issuecomment-375963696

In some previous cases of this issue, the RealSense user has been using a Virtual Machine (VM), such as **VMWare**.  Are you using a VM please?

jigglyjoubres, 2021-01-21:
Thank you for your response @MartyG-RealSense but no I am using a dual boot.

closed 2021-02-05 by MartyG-RealSense