realsenseai/librealsense #8204: T265 can't be recognized by realsense2.lib but RealSense Viewer works

swan-mojo-vision, 2021-01-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | 0.2.0.951| 
| Operating System & Version |   windows 10  | 
| Kernel Version (Linux Only)    |                             | 
| Platform                                 | PC  |
| SDK Version                            | WIN10-2.41.0.2657                       | 
| Language                            |  C++                         | 
| Segment			|  AR                         | 

### Issue Description
I built & ran the example "rs-pose.cpp" but only got:

RealSense error calling rs2_pipeline_start_with_config(pipe:000001B615C9F8A0, config:000001B615C9EA90):
    No device connected

apparently it got stuck at:              pipe.start(cfg);
also I tried:
rs2::context ctx;
std::cout<<ctx.query_devices().size();
but only got 0!

the device works just fine using "Intel RealSense Viewer".

I found/read through a similar issue report as in "Issue #6286", tried tweaks (e.g., using different SDK versions) from there but none of them works.

I'd greatly appreciate it if you could help resolving the issue.

Thanks

swan-mojo-vision, 2021-01-22:
Can anyone here take a look at this issue that I post two days ago? Thanks.

RealSenseSupport, 2021-01-28:
Hi,

We were able to run the rs-pose.cpp sample successfully on T265 with the latest firmware/SDK. Incidentally the only only time the exact error message that you are seeing occurs is when the T265 is not plugged in at all. It's possible you might be experiencing some USB connectivity issue?  

Thanks

swan-mojo-vision, 2021-02-10:
Thanks for the comments.
I tried it again using Visual Studio Community 2019, instead of using vscode where the reported issue occurred, and it works.
I'm wondering what's making it fail in vscode? The 'cmakelists.txt' I'm using is

`cmake_minimum_required(VERSION 3.8)
project(rs-pose)
add_executable (rs-pose rs-pose.cpp)
find_package(realsense2 CONFIG REQUIRED)
target_link_libraries(rs-pose PRIVATE realsense2::fw realsense2::realsense2 realsense2::realsense-file)`

The build with the above file succeeded, but returned error after launch:
`RealSense error calling rs2_pipeline_start_with_config(pipe:0000018D63B12FB0, config:0000018D63A75820):
    No device connected`

Thanks

ev-mp, 2021-02-10:
@swan-mojo-vision hello, 
Can you check that `-DBUILD_WITH_TM2=true` flag was set when compiling the lib? Additionally - follow cmake-generated log to verify that TM2's firmware image is downloaded (a possible network issue?).

swan-mojo-vision, 2021-02-10:
Hi @ev-mp,
Thanks for your response.
I don't think I did the realsense2 compilation separately - I just installed it by following the instructions: [](https://github.com/IntelRealSense/librealsense#building-librealsense---using-vcpkg)
and then build the `rs-pose.cpp` example in vscode using the `cmakelists.txt` as described earlier in this thread.

Am I missing something?

Thanks

ev-mp, 2021-02-10:
@swan-mojo-vision, please rebuild the demo with writing to log enabled (Debug verbosity) and post for review.
Additionally share the output of` rs-enumerate-devices` and `dmesg' kernel log

swan-mojo-vision, 2021-02-10:
@ev-mp , here is the build log:
`[main] Building folder: viSLAM 
[build] Starting build
[proc] Executing command: "C:/Program Files/cmake-3.19/cmake-3.19.0-rc1-win64-x64/bin/cmake.exe" --build c:/Users/swan/Box/ShengWan/viSLAM/build --config Release --target all -- -j 14
[build] "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -SC:\Users\swan\Box\ShengWan\viSLAM -BC:\Users\swan\Box\ShengWan\viSLAM\build --check-build-system CMakeFiles\Makefile.cmake 0
[build] "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E cmake_progress_start C:\Users\swan\Box\ShengWan\viSLAM\build\CMakeFiles C:\Users\swan\Box\ShengWan\viSLAM\build\\CMakeFiles\progress.marks
[build] C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/bin/mingw32-make.exe  -f CMakeFiles\Makefile2 all
[build] mingw32-make.exe[1]: Entering directory 'C:/Users/swan/Box/ShengWan/viSLAM/build'
[build] C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/bin/mingw32-make.exe  -f CMakeFiles\rs-pose.dir\build.make CMakeFiles/rs-pose.dir/depend
[build] mingw32-make.exe[2]: Entering directory 'C:/Users/swan/Box/ShengWan/viSLAM/build'
[build] "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E cmake_depends "MinGW Makefiles" C:\Users\swan\Box\ShengWan\viSLAM C:\Users\swan\Box\ShengWan\viSLAM C:\Users\swan\Box\ShengWan\viSLAM\build C:\Users\swan\Box\ShengWan\viSLAM\build C:\Users\swan\Box\ShengWan\viSLAM\build\CMakeFiles\rs-pose.dir\DependInfo.cmake --color=
[build] Scanning dependencies of target rs-pose
[build] mingw32-make.exe[2]: Leaving directory 'C:/Users/swan/Box/ShengWan/viSLAM/build'
[build] C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/bin/mingw32-make.exe  -f CMakeFiles\rs-pose.dir\build.make CMakeFiles/rs-pose.dir/build
[build] mingw32-make.exe[2]: Entering directory 'C:/Users/swan/Box/ShengWan/viSLAM/build'
[build] [ 50%] Building CXX object CMakeFiles/rs-pose.dir/rs-pose.cpp.obj
[build] C:\PROGRA~2\MICROS~3\2019\COMMUN~1\VC\Tools\MSVC\1428~1.293\bin\Hostx64\x64\cl.exe  /nologo /TP  -IC:\src\vcpkg\vcpkg\installed\x64-windows\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG /FoCMakeFiles\rs-pose.dir\rs-pose.cpp.obj /FdCMakeFiles\rs-pose.dir/ /FS -c C:\Users\swan\Box\ShengWan\viSLAM\rs-pose.cpp
[build] rs-pose.cpp
[build] [100%] Linking CXX executable rs-pose.exe
[build] "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E cmake_link_script CMakeFiles\rs-pose.dir\link.txt --verbose=1
[build] "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\rs-pose.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100183~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100183~1.0\x64\mt.exe --manifests -- C:\PROGRA~2\MICROS~3\2019\COMMUN~1\VC\Tools\MSVC\1428~1.293\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\rs-pose.dir\objects1.rsp  /out:rs-pose.exe /implib:rs-pose.lib /pdb:C:\Users\swan\Box\ShengWan\viSLAM\build\rs-pose.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:console  C:\src\vcpkg\vcpkg\installed\x64-windows\lib\fw.lib C:\src\vcpkg\vcpkg\installed\x64-windows\lib\realsense2.lib C:\src\vcpkg\vcpkg\installed\x64-windows\lib\realsense-file.lib C:\src\vcpkg\vcpkg\buildtrees\realsense2\x64-windows-rel\common\fw\fw.res kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib 
[build] C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -noprofile -executionpolicy Bypass -file C:/src/vcpkg/vcpkg/scripts/buildsystems/msbuild/applocal.ps1 -targetBinary C:/Users/swan/Box/ShengWan/viSLAM/build/rs-pose.exe -installedDir C:/src/vcpkg/vcpkg/installed/x64-windows/bin -OutVariable out
[build] mingw32-make.exe[2]: Leaving directory 'C:/Users/swan/Box/ShengWan/viSLAM/build'
[build] [100%] Built target rs-pose
[build] mingw32-make.exe[1]: Leaving directory 'C:/Users/swan/Box/ShengWan/viSLAM/build'
[build] "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E cmake_progress_start C:\Users\swan\Box\ShengWan\viSLAM\build\CMakeFiles 0
[build] Build finished with exit code 0
`

swan-mojo-vision, 2021-02-10:
@ev-mp , the output of `rs-enumerate-devices`:
`C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-enumerate-devices
Device info:
    Name                          :     Intel RealSense T265
    Serial Number                 :     11622110830
    Firmware Version              :     0.2.0.951
    Physical Port                 :     \\?\USB#VID_8087&PID_0B37#11622110830#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
    Product Id                    :     0B37
    Usb Type Descriptor           :     3.1
    Product Line                  :     T200

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format
    Fisheye 1     848x800       @ 30Hz     Y8
    Fisheye 2     848x800       @ 30Hz     Y8
    Gyro         N/A            @ 200Hz    MOTION_XYZ32F
    Accel        N/A            @ 62Hz     MOTION_XYZ32F
    Pose         N/A            @ 200Hz    6DOF`

ev-mp, 2021-02-11:
@swan-mojo-vision , all the logs suggest that the SDK is properly configured and that the camera is present.
But while I can see  `cmake build` stage there is no log for `cmake configure` and its flags.
I'd suggest to make a clean build and retry.
And if still not recognized - rebuild `rs-pose` with the SDK's log enabled:
```cpp
int main(int argc, char* argv[])
{
    log_to_file(RS2_LOG_SEVERITY_DEBUG, "librealsense.log");
    ...
```

swan-mojo-vision, 2021-02-11:
@ev-mp ,
Not sure if this is the log you were mentioning:
'
[main] Configuring folder: viSLAM 
[main] Configuring folder: viSLAM 
[driver] Removing c:/Users/swan/Box/ShengWan/viSLAM/build/CMakeCache.txt
[driver] Removing c:\Users\swan\Box\ShengWan\viSLAM\build\CMakeFiles
[proc] Executing command: "C:/Program Files/cmake-3.19/cmake-3.19.0-rc1-win64-x64/bin/cmake.exe" --no-warn-unused-cli -DCMAKE_TOOLCHAIN_FILE:STRING=C:\src\vcpkg\vcpkg\scripts\buildsystems\vcpkg.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Release -Hc:/Users/swan/Box/ShengWan/viSLAM -Bc:/Users/swan/Box/ShengWan/viSLAM/build -G "MinGW Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The C compiler identification is MSVC 19.28.29337.0
[cmake] -- The CXX compiler identification is MSVC 19.28.29337.0
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: C:/Users/swan/Box/ShengWan/viSLAM/build
'

ev-mp, 2021-02-12:
@swan-mojo-vision , we need to review the `cmake configure` Librealsense invoked with "cmake .. -DBUILD_...."
to check the configuration flags and the output

swan-mojo-vision, 2021-02-12:
@ev-mp Note sure where to find the  `cmake configure` as I'm using `cmakelists.txt` in `vscode`, and I'm not quite familiar with the cmake (and I'm not directly using it). Can you please advise?
Here again is the complete `cmakelists.txt` file:

```
cmake_minimum_required(VERSION 3.8)
project(rs-pose)
add_executable (rs-pose rs-pose.cpp)
find_package(realsense2 CONFIG REQUIRED)
target_link_libraries(rs-pose PRIVATE realsense2::fw realsense2::realsense2 realsense2::realsense-file)
set(CMAKE_VERBOSE_MAKEFILE ON)
```

With the `realsense2` installed by using `vcpkg`, it allows me to build and run the `rs-pose.cpp` ...
Am I missing something here?

swan-mojo-vision, 2021-02-12:
Also noticed there's a CMakeOutput.log file was produced, and I list it below (not sure if it helps):
```

The system is: Windows - 10.0.19042 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe 
Build flags: 
Id flags:  

The output was:
0
Microsoft (R) C/C++ Optimizing Compiler Version 19.28.29337 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

CMakeCCompilerId.c
Microsoft (R) Incremental Linker Version 14.28.29337.0
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:CMakeCCompilerId.exe 
CMakeCCompilerId.obj 

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.exe"

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.obj"

The C compiler identification is MSVC, found in "C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/3.19.0-rc1/CompilerIdC/CMakeCCompilerId.exe"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe 
Build flags: 
Id flags:  

The output was:
0
Microsoft (R) C/C++ Optimizing Compiler Version 19.28.29337 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

CMakeCXXCompilerId.cpp
Microsoft (R) Incremental Linker Version 14.28.29337.0
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:CMakeCXXCompilerId.exe 
CMakeCXXCompilerId.obj 

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.exe"

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.obj"

The CXX compiler identification is MSVC, found in "C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/3.19.0-rc1/CompilerIdCXX/CMakeCXXCompilerId.exe"

Detecting C compiler ABI info compiled with the following output:
Change Dir: C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/bin/mingw32-make.exe cmTC_80769/fast && C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/bin/mingw32-make.exe  -f CMakeFiles\cmTC_80769.dir\build.make CMakeFiles/cmTC_80769.dir/build

mingw32-make.exe[1]: Entering directory 'C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp'

Building C object CMakeFiles/cmTC_80769.dir/CMakeCCompilerABI.c.obj

C:\PROGRA~2\MICROS~3\2019\COMMUN~1\VC\Tools\MSVC\1428~1.293\bin\Hostx64\x64\cl.exe  /nologo   /DWIN32 /D_WINDOWS /W3  /MDd /Zi /Ob0 /Od /RTC1 /FoCMakeFiles\cmTC_80769.dir\CMakeCCompilerABI.c.obj /FdCMakeFiles\cmTC_80769.dir/ /FS -c "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\share\cmake-3.19\Modules\CMakeCCompilerABI.c"

CMakeCCompilerABI.c

Linking C executable cmTC_80769.exe

"C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E cmake_link_script CMakeFiles\cmTC_80769.dir\link.txt --verbose=1

"C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_80769.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100183~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100183~1.0\x64\mt.exe --manifests -- C:\PROGRA~2\MICROS~3\2019\COMMUN~1\VC\Tools\MSVC\1428~1.293\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\cmTC_80769.dir\objects1.rsp  /out:cmTC_80769.exe /implib:cmTC_80769.lib /pdb:C:\Users\swan\Box\ShengWan\viSLAM\build\CMakeFiles\CMakeTmp\cmTC_80769.pdb /version:0.0 /machine:x64  /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib 
C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -noprofile -executionpolicy Bypass -file C:/src/vcpkg/vcpkg/scripts/buildsystems/msbuild/applocal.ps1 -targetBinary C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp/cmTC_80769.exe -installedDir C:/src/vcpkg/vcpkg/installed/x64-windows/debug/bin -OutVariable out

mingw32-make.exe[1]: Leaving directory 'C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp'

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/bin/mingw32-make.exe cmTC_09045/fast && C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/bin/mingw32-make.exe  -f CMakeFiles\cmTC_09045.dir\build.make CMakeFiles/cmTC_09045.dir/build

mingw32-make.exe[1]: Entering directory 'C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp'

Building CXX object CMakeFiles/cmTC_09045.dir/CMakeCXXCompilerABI.cpp.obj

C:\PROGRA~2\MICROS~3\2019\COMMUN~1\VC\Tools\MSVC\1428~1.293\bin\Hostx64\x64\cl.exe  /nologo /TP   /DWIN32 /D_WINDOWS /W3 /GR /EHsc  /MDd /Zi /Ob0 /Od /RTC1 /FoCMakeFiles\cmTC_09045.dir\CMakeCXXCompilerABI.cpp.obj /FdCMakeFiles\cmTC_09045.dir/ /FS -c "C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\share\cmake-3.19\Modules\CMakeCXXCompilerABI.cpp"

CMakeCXXCompilerABI.cpp

Linking CXX executable cmTC_09045.exe

"C:\Program Files\cmake-3.19\cmake-3.19.0-rc1-win64-x64\bin\cmake.exe" -E cmake_link_script CMakeFiles\cmTC_09045.dir\link.txt --verbose=1

[... cut ...] lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib 
C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -noprofile -executionpolicy Bypass -file C:/src/vcpkg/vcpkg/scripts/buildsystems/msbuild/applocal.ps1 -targetBinary C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp/cmTC_09045.exe -installedDir C:/src/vcpkg/vcpkg/installed/x64-windows/debug/bin -OutVariable out

mingw32-make.exe[1]: Leaving directory 'C:/Users/swan/Box/ShengWan/viSLAM/build/CMakeFiles/CMakeTmp'

```

closed 2026-10-23 by MartyG-RealSense